githubdoe/DFTFringe #6: "show resized image" crashes application

atsju, 2022-11-20:
I load wat.jpg, autotrace outline, done, then "show resized image" crashed the application

version 6.0 on win10

githubdoe, 2022-11-21:
This is not the place to talk about problems you may have with the program.  Instead use the groups.io interferometry site.  At
https://groups.io/g/Interferometry.   I will look for messages from you there.

closed 2022-11-21 by githubdoe

atsju, 2022-11-21:
I'm firmware engineer and "in general", open source software take problems (I understand for the part "no question") on the issue tab in GitHub. If you want not issue here, maybe you should go to "settings" on github repo and untick the "issues" feature.

githubdoe, 2022-11-21:
Posting issues here is fine except I don't pay attention to it and find it easier to discusses on the other site.

atsju, 2022-11-21:
Not sure to fully understand, these are quite opposite messages. Anyway, I will post on groups.io if you prefer so. 

Some last "in general" or "industry standard" (please take no offense, I just do this professionally and also helped on some small open source repos like yours. The tool is great!): GH issues are closed when the code is updated with a fix, even better if the issue is linked to a pull request fixing the actual code. Pull requests and commit with small changes (with peer code review) are really recommended. Versions number for releases should follow https://semver.org/