githubdoe/DFTFringe #17: when averaging wavefronts, ignore regions should be dealt with better

gr5, 2023-03-17:
example data and result:
https://groups.io/g/Interferometry/topic/97668823#33021

I'd like to work on this issue.  It probably won't be for a few weeks but I'd like to look at the code for this and see if I can come up with a solution.

gr5, 2023-03-17:
Thinking about this more, it's quite complicated!  The wavefronts displayed are not the wavefronts stored.  The stored wavefronts have HUGE amounts of tilt and defocus and each of the wavefronts to be averaged may have very different values for (for example) tilt.  So when you include all the data in 2 wavefronts and average them together the new wavefront will have a different tilt from either of the other two.

If you skip a few points because they are ignored, they won't get the proper tilt and neighboring pixels, one pixel a proper average will give a very different value from one pixel over which only uses one of the wavefronts.

So for the "ignore" pixels we still need to average in *something*.  
1) Maybe the nearest non-ignored pixel?  If it's a large ignore region that could be pretty far away.  
2) So maybe the unchecked zernicke's should be used to calculate the best fit replacement pixel.  No I think it should use piston,tilt,defocus every time.  Because 2 different igrams are *expected* to have different values for these 4 terms even in a perfect world.  whereas coma, etc shouldn't change from one igram to the next.

I think option #2 is best.  The "ignore" areas get the pixel value for a simulated wavefront with the same piston,tilt,defocus.

githubdoe, 2023-03-18:
I'm surprised that I don't average the nulled wave fronts instead of the raw (.Data) member.  Perhaps there is a reason that I no longer remember.  But I would have you try using the .nulled wavefront member instead in the average routine in surfaceManager.

githubdoe, 2023-03-18:
The problem with using the .nulled member is that then the null goes away in the final result.  One could add the null back into the result.

githubdoe, 2023-03-18:
The other option would be to null with only defocus and tilt.   The nulling routine can be passed what zernike values should be nullded so this could be done to each wavefront just before add it to the average.

gr5, 2023-07-28:
Fixed in PR 57

closed 2023-07-29 by gr5