githubdoe/DFTFringe #19: [bug] autoload outline won't load exclusion region

atsju, 2023-03-26:
Version 6.2

From my understanding, today the .oln file contains the outline, regions and the DFT limit. This lets user reload same picture and modify slightly outline. When loading igram, the outline is loaded automagically.

Problem: on my computer, when loading an igram, the corresponding outline and DFT are loaded, but regions are not.

gr5, 2023-07-29:
Looking at this now.

githubdoe, 2023-07-29:
I think that is because the regions were not saved properly into the QSettings when they are modified at crop time.
I think I fixed that IIRC recently in the last few weeks if my changes got pushed.

gr5, 2023-07-30:
I thought *I* fixed this a few weeks ago, lol.

I wonder if we both made changes that were incompatible and it's broken again.  Anyway I will look at this next.  It's definitely not 100% working as expected.

githubdoe, 2023-07-30:
All I did if it made it into the archive was to add saving them to the QSettings when deleted or crop was invoked. IIRC  however that was around the time my disk drive crashed.  Then what I remember is trying to recreate the fix and then pushing it.  So it might be with my last push which also delt with the config edge mask.

PR #63 (2023-08-03): Fixes GUI issues related to outline (oln) file (stores inner, outer, regions)
Fixes #19  and more:
- Fixed bug where oln files can be read wrong (typically 12% of the time) and
  the fix involved creating a new json based oln format.  Old oln file formats
  are detected and are still read using the "older" oln reading code.
- Other outline and region related GUI bugs
- Fixed bug where reading and writing to oln file after cropping was using
  the wrong x,y values.  Works great now.
- Fixed minor bug recently introduced where it stopped showing the version 
  number in the metrics window (top o [... cut]
closed 2023-08-03 by gr5