githubdoe/DFTFringe #22: [Feature request] crash report for people to help report

atsju, 2023-03-26:
I had 2 crashes while using official version 6.2

I have been unable to reproduce the crashes reliably so I can't even tell you what to fix. I think DFTF should catch all exceptions and either write into a log file or show a crash window for people to copy paste the call stack into an issue. It should not close/crash silently. This is standard practice to help developpers fix issues happening on "customer" side.

here are information from windows event report. Not very usefull.
```
Nom de l’application défaillante DFTFringe.exe, version : 0.0.0.0, horodatage : 0x63db083b
Nom du module défaillant : libgcc_s_seh-1.dll, version : 0.0.0.0, horodatage : 0x00000000
Code d’exception : 0x40000015
Décalage d’erreur : 0x000000000000d6a5
ID du processus défaillant : 0x3b8c
Heure de début de l’application défaillante : 0x01d95ff7eda6c2ff
Chemin d’accès de l’application défaillante : C:\DFTFringe64\DFTFringe.exe
Chemin d’accès du module défaillant: C:\DFTFringe64\libgcc_s_seh-1.dll
ID de rapport : 18c77f35-82bc-4ac2-8d20-1c7e59baf140
Nom complet du package défaillant : 
ID de l’application relative au package défaillant : 
```
```
Nom de l’application défaillante DFTFringe.exe, version : 0.0.0.0, horodatage : 0x63db083b
Nom du module défaillant : libopencv_core3412.dll, version : 0.0.0.0, horodatage : 0x5fddbfdb
Code d’exception : 0x40000015
Décalage d’erreur : 0x000000000023e078
ID du processus défaillant : 0x39d8
Heure de début de l’application défaillante : 0x01d95ff0660379b8
Chemin d’accès de l’application défaillante : C:\DFTFringe64\DFTFringe.exe
Chemin d’accès du module défaillant: C:\DFTFringe64\libopencv_core3412.dll
ID de rapport : a8ff05b6-894e-4a0f-b156-5b5ec77b43d4
Nom complet du package défaillant : 
ID de l’application relative au package défaillant : 
```

gr5, 2023-08-03:
I was thinking a log file.  Currently we write to qDebug() output which only runs if you compile debug version and that output goes to some special console (not stdout).  Instead I'd like some kind of a log file.  Crash info could be dumped into the same file.  It should include dates and times and a reliable line at the start whenever you relaunch the program.

A log file can help debug things that don't cause crashes as well.  The help menu in DFTF could have an entry to show you the folder where the log file can be found.  And even try to open the folder with the file highlighted maybe?  Or copy the file to the paste buffer maybe?

atsju, 2023-08-03:
100% agree with a log file/folder.
Note the file should just not grow indefinitely but I'm sure there are some standard things to manage logs. Let's not reinvent the wheel.
Opening the path to folder of log file should be enough.

githubdoe, 2023-08-03:
I remember there is a lot of info about qDebug and how to create or modify it to create a log file.  It was way more complicated than I wanted to spend on it at the time.  However there may be solutions already documented somewhere on how.

atsju, 2023-08-09:
I asked around at work about logging.
There are some logging libraries listed at cppreference https://en.cppreference.com/w/cpp/links/libs under "Logging". This is THE C++ reference website so I think it's good to choose one listed there.
In my team they use https://github.com/gabime/spdlog. It's cross platform and a good candidate IMHO with almost 20k github stars.

gr5, 2023-08-09:
I'm not decided for sure but I thought using qDebug would be nice since we already log lots of stuff using that.  I didn't look into it though.

If we are using 3rd party logging then it would be nice if it included date and time in the log.

atsju, 2023-08-09:
> I'm not decided for sure but I thought using qDebug would be nice since we already log lots of stuff using that. I didn't look into it though.
> 
> If we are using 3rd party logging then it would be nice if it included date and time in the log.

There are 199 `Qdebug` occurrences in the code only. I feel like these would probably be faster to change than implementing a logger from `Qdebug` itself. 
Output will of course have a date and also a log level like:
`[2015-08-24 07:29:04.502] [app_logger] [info] Log from application`

```
logger.set_level(spdlog::level::debug);
logger.warn("this should appear in both console and file");
logger.info("this message should not appear in the console, only in the file");
```
You have logging levels, each message has a level so that you can choose at runtime if you want it always or only for very verbose runs.

I don't know Qdebug, so if you think it's easy why not but you would lose the log levels

githubdoe, 2023-08-09:
The thing I use QDebug for is to print debug stuff to the console when in the Qt IDE.  That feature must stay available.  That is a log must print to the app output console as well as log to a file if desired.

The reason this is important to me is there are times when  the debugger fails when a breakpoint is set in a routine that has cv::Mats in it.  (a lot of DFTFringe functions do.)  Very frustrating.  In that case I can only get access to the variables during debug by disabling the break point and then by having it print them to the output.   Of course that output is only available when in the IDE and perhaps also in Debug build mode.

There are discussions in the qt forums about how to do that I think.  Might be mixing it up with the loggers that I used with python.

atsju, 2023-08-09:
The logger I propose is very near the python logger (which I know better). You should be able to print to console and/or to file. And this can be depending or independant of log level. I think it fits your need. 
This is only a proposal and open for discussion for now. I did not work on it and I think so does George. 

`Qdebug << "message";` will become `logger.info("message");` or `logger.debug("message");` for example.

githubdoe, 2023-08-09:
The great thing about QDebug is it's ability to log any Qt object without any format statement or fancy routine to get to each important element in the class being printed.   All done with a simple << operator.

atsju, 2023-08-09:
Yes it is great as a print statement. But I think there is a slight different between a print and a logger. 
If we want to stay with QT maybe https://doc.qt.io/qt-6/qmessagelogger.html is a better start.
There is nothing wrong debugging with Qdebug, but code sent to end user uses logger in general so that you can easily enable/disable every log/print

githubdoe, 2023-08-09:
The problem of converting every Qdebug to a logging statement is that many are left over from debug sessions that should not be logged.  But because of the way debug and release work they don't hurt the real app.  However they could put a burden on the log file.  I bet there are a lot of superfluous debug statements currently.  In fact all probably are.  However when I get into debug mode I find they can still be useful to help me track code execution.  Especially when QDB dies like it does most of the time.  It is nice to not have to go back and insert them.

gr5, 2023-08-09:
Okay so it sounds like we keep qDebug as it is and add a new logging system for "user logging".  For issues that are hard to duplicate (of which I have one right now - it happens so damn rarely and I can't duplicate it).

The advantage of qDebug is that it only logs if you compile debug mode with the DEBUG flag.

The other logging is for "user" debugging.

Also I noticed that the logging system Julien found supports dumping QT objects.  At least some of them.

githubdoe, 2023-08-09:
So perhaps separate debug from logging.  Don't try to retrofit the old debug statements.  As you can probably tell I have no idea how to make or use a crash dump.  One could be useful if you know how I expect.

gr5, 2023-08-09:
It depends on the crash dump type but it can be incredibly useful where it can tell you the whole call stack and show which function and which line of code failed and what the error was (usually related to accessing memory that it can't).

I don't know if it can tell you all that from a release version or if you need the debug version to get function names and line of code.

I don't know what utilities you need or how to use them to load the dump and analyze it.  But I've done it before.  I think maybe in Linux.

atsju, 2023-08-09:
One last word about Qdebug VS logger.
We could easily convert Qdebug to logger.debug which is lowest level. End user would probably log only warning level by default so those would have no impact but still would be printed if user sets level to maximum verbosity. Well this is how I would do personally but keeping both Qdebug and a logger in parallel is acceptable solution.

About cryitcal crashlog, I think it's just a try catch all that get pushed into the logging system. It will be easy to generate a crash to see what kind of information are available in release once we have the logging feature. I agree crashlog can be incredibly usefull if there are enough information.

atsju, 2023-08-17:
I'm currently documenting myself on how to add the logging feature and crash dump. I will probably work on it before going back to memory leaks.

You might be interested that there was no cross-platform way to get the call stack (typically for crash log). But this was until C++23 ! => https://en.cppreference.com/w/cpp/utility/basic_stacktrace

I will stick to C++11 for now so maybe we won't have call-stack for every platform. At least for the beginning.
C++17 if we move to QT6 in 2025/2026 or so (end of commercial support).

atsju, 2023-08-17:
OK crash report and logger are 2 separate topics. I now work on logger.

I did not try on windows for the moment but current state is : 
- I'm using spdlog. It works really well and chatgpt helped me set it up. I'm actually impressed, less than 2h reading and 2h coding and 10 lines to get tuned basics to work
- logger prints to console when executed in command line, else no print there (to be confirmed on windows)
- logger always prints same information to a rotating buffer of log file

Here is current format (don't mind the actual message) :
[image]
or
[image]

Questions :  

1. Do we want this format or slightly different one ? I can do almost anything
2. Do we want to get file and line in format ? 
3. There are 6 levels (trace, debug, info, warn ,error, critical) I should be able to make them settable with env variables. I propose to set "info" or "warn" by default. Us developers or end users experiencing problems should/may set "trace" on computer to get more logs. Does this sound OK for you ?
4. Are you OK with the console + rotating files logging ? There are plenty of solutions

atsju, 2023-08-17:
I confirm on windows it's like for other prints. Launching from "cmd" console won't log anything. However using QTcreator will get the messages in it's console if launched from there. I think this is compatible with our use and we still have the file. Just let me know.

[image]

I will finish implementation when you confirm all open questions. Also I recommend this comes later in v7.1.0, this way we test it a bit at home before shipping

gr5, 2023-08-17:
don't want line number.  First line output to logger should be obvious ("app started" or something similar).  Include version information at startup.  Maybe a line of code (at startup) that shows free memory?

Explain a little more about the "trace" feature please.

When you said stack trace doesn't work on all platforms - most important is windows as that is what most people are using and where I debug.

Somewhere in DFTF there should be a message showing where log file is located (maybe "help" "about"?)

githubdoe, 2023-08-17:
Years ago when trouble shooting out of memory issues I discovered it is complex to get windows memory usage.  There may still be some routines in DFTFringe to get that info somewhere probably in Mainwindow or even in Main.

githubdoe, 2023-08-17:
Stack trace is what I use to find where it crashed.  So that is the most important info if available.  However that usually is not available in release builds because release exes do not have the code to line number info in them.  Perhaps there are modern solutions but I don't know.

githubdoe, 2023-08-17:
I think log in the install is just fine.

githubdoe, 2023-08-17:
There can be times when QDebug is not desired to output because it is in a tight loop and printing slows down the execution tremendously.  Thus I would not expect a logger for the released version to print any of the QDebug stuff.  That is one of the things I found useful about the current QDebug functionality.  Did not have to worry about any unintended debug messages left over.

gr5, 2023-08-17:
I wrote a reply and didn't finish it and now it seems to be gone.

I prefer no line numbers please.  Each message should be unique enough that we can search for it and should be unique.

First line logged on startup should include version information and should be obvious (maybe blank line (except for date/time/etc) just before).

I like the idea of rotating with size around 1MB but don't feel strongly about this feature either way.

It should be easy for user to find log file.  e.g. under "help" "about".

Please explain more about "trace".  Log levels should default as you said but controllable in the config menu system.

You mentioned also the ability to dump stack trace on crash, right?  You said only certain platforms do that?  Does that work on windows as that's really the only one I care about since most (all?) users use windows at the moment.

gr5, 2023-08-17:
Oh - my reply is WAY WAY up there!  I didn't realize there were so many messages.  Sorry for the repeats.  New point:

Often installation folder is write protected from user and only can be written by admin.  So you either need a backup location or maybe use %appdata%/DFTFringe/ folder which in windows is one of the easy-to-get folders.  I think there might be a way to find a safe "user" folder to write to in QT.

atsju, 2023-08-17:
OK thanks. 

> You mentioned also the ability to dump stack trace on crash, right? You said only certain platforms do that? Does that work on windows as that's really the only one I care about since most (all?) users use windows at the moment.

Yes it's what I WANT. It should be doable but I don't know right now.
And as already stated : 
> I will look a it later. It's not that it does not work on all platforms, it's that there in no "one fit all" solution to implement it (except with C++23)

The way the pre-C++23 tools do is that they look the stack, and search for stacked return addresses to deduce function calls that happened before the crash. I didn't find any library doing this for ALL platforms and I understand I should look mainly for windows.

OK I wasn't sure you lost your message way up so here is a more detailed answer anyway