githubdoe/DFTFringe #23: file open dialog for igrams needs to include uppercase JPG

gr5, 2023-04-25:
When opening an igram among the list of extensions (*.jpg) should also be included "*.JPG" (and *.JPEG, *.jpeg, and perhaps png and perhaps more).

In linux file names are case sensitive.  You can have a.jpg and a.JPG and they are two separate files potentially in the same folder.  You can't do that in windows.  Adding "JPG" will make the program easier to use (you don't have to choose "all file types" and you don't have to avoid all the other non-image files).

atsju, 2023-10-15:
There is a misuse in code between https://doc.qt.io/qt-5/qfiledialog.html#setNameFilters and https://doc.qt.io/qt-5/qfiledialog.html#setMimeTypeFilters

This is bothering me when working on linux for anything related to memleaks. I will work on it

atsju, 2023-10-15:
also linked to https://bugreports.qt.io/browse/QTBUG-51712

atsju, 2023-10-15:
The https://doc.qt.io/qt-5/qfiledialog.html#setMimeTypeFilters is only a a shortcut for easy name filters.
[image]
`("All files (*)", "image Windows BMP (*.bmp *.dib)", "image GIF (*.gif)", "image JPEG (*.jpg *.jpeg *.jpe)", "image PNG (*.png)", "image SVG (*.svg)", "image SVG compressée (*.svgz)", "icône Windows (*.ico)", "image PBM (*.pbm)", "image PGM (*.pgm)", "image PPM (*.ppm)", "image XBM (*.xbm)", "image XPM (*.xpm)")`

I propose to put every relevant extension in image manually and it will display like this :
[image]

Not the most elegant way from QT point of view but at least it works.
@githubdoe @gr5  Please validate that it's OK all types are in same line and we don't need a drop down. Note that people can use dialog "sort" feature if they realy mix different picture types in same folder. I will open PR soon.

I could do the drop down from original supported mime types and add uppercase (I even started to write the code at some point) but it's not worse the effort. Only difficult to maintain.

PR #103 (2023-10-16): fix uppercase JPG not showing in linux
fix #23
closed 2023-10-16 by gr5