githubdoe/DFTFringe #24: Changes to get dftfringe to compile for a mac

gr5, 2023-05-29:
This is a placeholder to discuss any changes needed to get DFTFringe to compile on a mac.

atsju, 2023-05-29:
(Link to groups for historical reasons https://groups.io/g/Interferometry/topic/99192778) 

I would like to add the CI for macOS. 

I also planned to add CD for supported platforms but my CI proposals need to be merged firsts. Is there any plan to merge them ? Or are more explanations needed ? It's too disruptive ?

githubdoe, 2023-05-29:
George you can take over and do any merging needed.  I don't have any updates that I remember and am very busy with other things.  Do you need any additional permissions?
 
Dale

gr5, 2023-05-29:
I will do that then.  I should be all set.

Interferometry, 2023-07-06:
Ive now uploaded a build of the macOS version here:

https://groups.io/g/Interferometry/files/DFTFringe%20for%20macOS

Consider it a beta, feedback welcome!

I'll post the code changes shortly. This is going to be really hard to do the CI for, this app bundle was very 'handmade'!

atsju, 2023-07-06:
Hi Dan, 
I don't own a Mac, however I did setup the CI for windows and linux on GitHub.  I now have a good view of the build process AND packaging with QT installer (which I choose because it's supposed to be multi-platform and support Mac).
I can help you setting up CI for Mac, do not hesitate to write some notes about how you built it on your own machine and it should be doable on CI.
Do not wait for perfection before pushing and opening a PR. It's better to discuss small changes along the road than having everything very rigid at the end of the process.
Thank you for your contribution !

PS: I deleted my answer on groups.io to paste it here as it's about development.

gr5, 2023-07-06:
The way to solve "hard" is to take it one step at a time.  Code changes are probably needed first.

I believe there are at least 2 very common and active versions of mac software?  Using different hardware technology?  I'm not sure - I just hear crazy stories from other multi  OS projects.

gr5, 2023-07-06:
I see you have libarmadillo.11.dylib in that zip file you uploaded.  I'm pretty sure it's not needed.  Curious if it still runs without that library.

Dale claims he only uses armadillo for one structure.  He was planning to use much more but changed his mind.

The armadillo.dll file is not included in Dale's installer even though it is mentioned in the ".pro" file.  I think @atsju can remove it from his installer as well but was going to test that before I told him that.  Don't think we need to build armadillo either unless the build process is creating the .h files (sometimes you have to build the include files which is more of a copy operation really).