githubdoe/DFTFringe #25: ignore regions sometimes stop being used

gr5, 2023-05-29:
I load igram DSC_1452.  There should already be an ignore region around the bad spot from oln file.  click done.  Click computer surface.  All works well every time.

Now change a zernike checkbox and click "recompute".  It seemingly reanalyzes the ignore region and fills it in with crazy values - many waves up/down from normal.

atsju, 2023-05-29:
You should probably add DFTFringe version (and OS + build type) where you observe the issue. Except if you are confident to fix it soon, experience tells me that it's a key information in long term if we need to reproduce.

Maybe we should have template ? Or is groups.io still prefered for the moment for reporting problems. https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/configuring-issue-templates-for-your-repository

gr5, 2023-05-29:
I plan to fix this sometime.  I just wanted this issue here so I don't forget it.  I used version 6.2 found on github.  "/releases/" folder.

gr5, 2023-05-29:
I think I know what the bug is as I was looking at this code today.  There is a function called fillVoid() and the way DFTF decides to call it or not is questionable.  I think it avoids calling this when you first analyze the igram but it can get called (and it shouldn't) when you recalculate.  Maybe.  I'm not certain but it's a good place to check first.  

On second thought maybe it's something totally different.  Maybe an issue where unwrap is called a second time.

gr5, 2023-08-08:
This is fixed now.  I tested it just now with master branch.  I don't know which PR fixed it but I did at least 2 fixes of code related to this issue.  Will be fixed for sure in 7.0.0 (if not already fixed in 7.6.2)

closed 2023-08-08 by gr5