githubdoe/DFTFringe #26: Missing files in github

gr5, 2023-05-30:
I realize there are several dozen files missing in this repository.  I'll add them.  None are crucial.  Some are jpg files needed for the help documentation (html files) and some are color maps.  Nothing critical.  I'll take care of this.  This issue is to remind me to rectify this.

gr5, 2023-05-30:
To be a little more precise, there are 
5 ".cmp" (color map) files in the /ColorMaps/ folder
37 web page files in the /res/Help/ folder
3 files in /res/ folder (a.avi, wat.jpg, wats.png).

We should probably skip the a.avi file as nothing links to it (I did a grep through all files in the repository) and it shows a very old version of DFTFringe in the video and there are plenty of videos out there on youtube explaining DFTFringe.

All the other files should be added to the repository.  To locate the files just install DFTF version 6.2.

atsju, 2023-06-26:
Hi @gr5 @githubdoe . 
I'm taking care of this issue right now. I have question about colormaps.

Colormaps present in release 6.2 but not in GitHub and not listed in .pro file are :

- Dale2.cmp 
- dalesColormap.cmp 
- daleverigated.cmp 
- EasonFav.cmp 
- xxx.cmp

Can you please confirm if these are Dales local favorites that have been added in release by error (and thus we don't add them to GitHub and futur releases)
or if these are colormaps highly optimized by Dale for DFTFringe usage and supposed to be exposed to the world ?

atsju, 2023-06-26:
About 3 files in /res/ folder (a.avi, wat.jpg, wats.png). => these seem to be unused. I won't add them

As for help, present in 6.2 but absent in repo are:
- res/Help/FailedAstigRemovalStandCountours.png
- res/Help/Forced.png
- res/Help/ForcedWithErrors.png
- res/Help/LightExposure.png => unused
- res/Help/LowExposure.png
- res/Help/Mirroutline.jpg => unused
- res/Help/QuickStart.html
- res/Help/Red.png => unused
- res/Help/StandAstigHelp.html
- res/Help/Thumbs.db => unused, probably not to be added
- res/Help/autoOutlineDisplays.png
- res/Help/autoOutlineOffsets.png
- res/Help/autooutline.png
- res/Help/autooutlineButton.png
- res/Help/blue.png => unused
- res/Help/darkExposure.png => unused
- res/Help/enlarged.png => unused
- res/Help/goodPeak.png
- res/Help/green.png => unused
- res/Help/greenIgram.png => unused
- res/Help/greenIgramsmall.png => unused
- res/Help/highExposure.png
- res/Help/intensityGraph.png
- res/Help/outlineControls.png => unused
- res/Help/outlineDebug.png
- res/Help/outlineing.html
- res/Help/overexposedgreen.png
- res/Help/pixelScanRange.png
- res/Help/printThru.png => unused
- res/Help/printthruProfile.png => unused
- res/Help/sizebars.png => unused
- res/Help/sml.png => unused
- res/Help/unwrapLine.png
- res/Help/unwrapLineAnalysis.png
- res/Help/useExsistingOutline.png
- res/Help/wellexposed.png
- res/Help/whiteDisk.png

I won't add the unused ones.

Note, all the pnj files present in repo are used.

githubdoe, 2023-06-26:
I don't remember if those were once part of the help system and had links to them or I was planning to add links to them.

githubdoe, 2023-06-26:
I have now figured out how to load a list of color maps.  If I had a hard time I imagine others have not succeeded.  I also figured out why I had trouble finding this issue list.  My default link to github takes me to skysolve and not DFTFringe.  I should have realized that but I did not find this topic.  Oh.   Sorry.

Like I told George.  I settled on Tropical but just out of laziness.  Now after looking at those with Dale in the name I like them.  It does not hurt to add them or any of the missing ones.  In reality most people want to use the program and not play with colors.

gr5, 2023-06-27:
Let's drop xxx.cmp and keep all the other cmp files.

atsju, 2023-06-27:
I will rename them Dale1 to dale4 for consistency if you are ok. 

And yes I never changed the default Colormaps once during my use.

githubdoe, 2023-06-27:
Fine with me.  It is too bad that I used the default color scheme to match OpenFringe.  Well that was necessary at the time so all those folks could compare the results of OpenFringe to the new DFTFringe with the same color scheme.  Now if we change the default color to one that I think give better info all will probably be confused.  Too bad.  I did a lot of work on color maps that probably was wasted.  Well it was fun for me so not a waste entirely.

Perhaps we could add a tip dialog that pops up at sometime asking if they would like a better color scheme and if yes give them a new one.  Then again is it worth it?  Maybe just add a video about how to get a better color scheme?  Also maybe not worth the trouble.  You guys can decide.

atsju, 2023-06-27:
Maybe next "big" release could simply change the color scheme to your optimized one. And we do a small tutorial on how to use original one again. If it's described as an improvement, people will try it and decide it's better for them too.

I would not do a popup. First because popups in tools are really annoying for user and second because this is one more thing to maintain. It's better to keep thing simple IMHO.

Kindly let me know which one is your preferred one so I name it "Dale1"

githubdoe, 2023-06-27:
I like the idea of doing it in a release and then include  how to change back in the release notes.   Even better is if the installer can ask if it ok to change color maps to new improved colors.

 It has been way too long and I no longer remember what I liked.  You guys try them and see what you like.

githubdoe, 2023-06-27:
Yes I think at one time I remember trying to name it better and settled on DalesColorMap.  So all the ones with Dale in the name could be reduced to just DalesColorMap.  At the time I got distracted from that task and they got left but included by my install build procedure.

Diverging maps are the most recommended by scientific experts on the use of color maps.  I like to see low spots as blue and high as red with green in the middle which is close to the Jet or Tropic but those can give similar colors for two diverse values.

gr5, 2023-06-27:
I think the "tropical" one is the most pleasing to look at and it is functionally quite useful as well.  I would vote to make "tropical" the default.  :)

If you want to change the default color - one option is to change it only for new users who don't have anything stored in their registry yet because they have never run DFTF before.

atsju, 2023-06-27:
I added 3 dales .cmp to #32 however I did not change the default one. I let you mange this in a future PR

PR #32 (2023-07-01): artifact with installer for windows
In this PR I will use QT installer framework to make installers. I use this instead of historial NSIS because it is multi-platform.
good link to learn : https://www.youtube.com/watch?v=1pKMcwJZay4

### Before merge I need:
- [x] installer for windows
- [x] use windeployqt to list and copy dependencies instead manual work
- [x] fix #26 

### do this separate to keep small steps
- [ ] installer for linux
- [ ] check which licences would need to be prompted (I suppose none is required for end user. Double check is required) or added in installation folder

###  [... cut]
commit 6ce54189 (2023-07-01): fix #26 add missing files in res\ folder
closed 2023-07-01 by gr5