githubdoe/DFTFringe #29: Check and potentially remove Armadillo usage

atsju, 2023-05-31:
https://github.com/githubdoe/DFTFringe/pull/18#pullrequestreview-1453584023

maybe we can remove it if unused

githubdoe, 2023-05-31:
It is used by zernike process for the code that Mike Peck gave me.   His code is used or is planned to be used to compute zernike values for annular interferograms values.

gr5, 2023-05-31:
okay.  Sounds good.  closing this issue.

closed 2023-05-31 by gr5

atsju, 2024-01-22:
Just to mention the issue has already been opened and closed. I think it must be linked with #119 

If we indeed update Mikes code then we need to keep Armadillo build. Else, we might remove the DLL and keep only header files. To be double checked

githubdoe, 2024-01-22:
I will try without the .dll just to see.

githubdoe, 2024-01-22:
It looks like I too do not have to use the armadillo.dll file and in fact do not have it in the release build. 
Code compiles with the latest downloaded armadillo.  Code crashes and will not start but that will be found.