githubdoe/DFTFringe #34: [BUG] colormap dynamic loading issue

atsju, 2023-06-27:
**Describe the bug**
v6.2
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to tool => user colormap
2. choose a map and Click on apply
3. right pane uses dynamically the new colormap but left one needs DFTF to restart first to load new config

**Expected behavior**
both panes should load the new colormap

**Desktop (please complete the following information):**
 - OS: windows10
 - DFTFringe version: 6.2

githubdoe, 2023-06-27:
You don't need to restart DFTFringe but that is one way.   The other is to switch wave fronts or re compute the wave front.

githubdoe, 2023-06-27:
Last night I got the 3D map to update to the new color by selecting a different wave front.  But not today.  Hmmm.

gr5, 2023-08-13:
There are workarounds - I think if you hit the "recompute" button above the zernike's it switches to the new colors.  There are other tricks to get the left one to use the new colors.  I agree this is a bug.

Also I think if you switch from "user color map" to one of the 3 or so built in ones, then switch back, it also fixes the left one.  Maybe.