githubdoe/DFTFringe #36: Regions come back after deletion.

githubdoe, 2023-06-29:
Worked on a project that had regions.

Switched to a different project and started to outline igram.  Regions from other project were there.  So I deleted them.  Then I auto traced the outline.  Regions came back.

So two issues. 

 Need some way to decide when to drop the saved regions that are saved in the QSettings.  Lets say on loading a different config or a different base path for igrams.

Figure out how deleted regions don't stay deleted.

githubdoe, 2023-06-29:
Yes the regions are not deleted from the QSettings.  That should be done in the delete outline function under the polyarea condition.  Deleting from the settings is not simple and perhaps the code to do so should be a subroutine that can be used also by the code that creates regions in the setting in the first place.

githubdoe, 2023-06-29:
In my opinion this is a bug that should have a high priority to be fixed and a release made.  Especially to help Walt with his chipped blank.  I will work on it and submit a pull request once I have it.

githubdoe, 2023-06-29:
I have fixed and tested the code.  Perhaps not all paths.  Next I need to figure out how to update the master here.  My version of the source has a bunch of changes that I have been playing with so I only want to update a couple files and not the whole source.  I once again have to learn the github magic incantations to do that.  Going back to Hogwarts to learn how.

atsju, 2023-06-30:
Hi @githubdoe. There are probably plenty of ways to do what you need using git. 
Do not hesitate to send me an email if you like to get some live guidance during a shared screen meeting or even if you prefer to explain your case and get a written guidance. If you don't want/need my help that's totally fine also.

githubdoe, 2023-07-01:
My laptop where the fix was has died.  I hope to be able to recover it but this is going to take time.

gr5, 2023-07-18:
I will fix this today.  Workaround is to hit the "crop" button after deleting or altering the regions.  The "done" button doesn't do it.  In the new version the regions will get deleted from registry when you click "done".

I could make the regions get deleted in the registry one by one as you delete them but I prefer to do it in "done".

Also I'll be fixing other bugs causing confusion and crashing so I think it will be fine to only update registry when you click "done".

githubdoe, 2023-07-18:
I never recovered my code fix (from the drive crash) that updated the registry when a region was deleted.  That was one place I discovered.  I think George in PM to me has discovered a few other cases.

gr5, 2023-07-18:
I fixed this seconds ago on my build.  I hope to have all the changes done today.

gr5, 2023-07-18:
On my build at the moment this is fixed but maybe not in the preferred way.  Right now you have to also click "done" to save the regions.  If you don't hit "done" you can recover the previous regions in the previous OLN file or from the previous igram by restarting DFTF or by just loading the igram again without hitting "done" first.

Is that okay?  I figured this way if you screw up you have a way to recover - if you didn't really want to delete the region.  This way it's not saved until you hit "done" to go to the next step.

This is my preferred UI but if you guys want it to completely delete the region out of the registry and also out of the OLN file "live" as you make the changes, that's pretty easy to do as well.

githubdoe, 2023-07-18:
I prefer to only update when "Done" is pressed.  Yes that was so one could abort the current op and recover.

gr5, 2023-07-18:
Good.  I have used this "feature" many times where I messed up the outline and just reloaded the igram.

PR #47 (2023-07-20): Fixes some issues with regions that were causing DFTF to crash
Fixes #36 

regions were scaled wrong when saved to registry (config memory - different from OLN files) such that they would typically go way off the left edge of the igram and not be visible to users and cause crashes.

Also fixed a bug where regions would be changed to their encompassing square.  But only in debug mode so didn't affect users.

Also changed it so regions are saved to registry (config) when you hit "done" and not when you hit "crop".  This way if the user makes a mistake with a region or other outline they can just [... cut]
closed 2023-07-20 by gr5