githubdoe/DFTFringe #37: null value calculated wrong if "edge mask" is checked and you restart dftf

gr5, 2023-07-01:
**Describe the bug**
null value sometimes ignores edge mask
I personally verified this bug found by Walt

**To Reproduce**
1) go to mirror config
2) disable edge mask and note (write down) null value
3) enable edge mask (use value of 1) and not new null value
4) exist, restart dftf
5) edge mask value is correct but null value is based on no edge mask
6) analyze an igram - note that it uses the wrong null value

**Screenshots**
Can be found in this discussion
https://groups.io/g/Interferometry/message/33334

githubdoe, 2023-07-01:
I have lost my D drive and am recovering.   It is a 2.5 inch Sata Samsung 1 TB SSD.   Getting a new one today at Best Buy.  Once recovered I can then recreate the work I did for the regions and fix this as well.  Hoping then to create a new release.  

I will start by using my jan 1 2021 last saved/backed up github. Hoping that this will have all my saved magic keys to enable github updates.  If not then it may take me longer to recreate the workable version.

Hoping against hope that if I connect the old D drive as external USB I can recover some data.  Need the Sata to USB adapter cable also picking up from Best Buy today.

githubdoe, 2023-07-01:
Oh last backup was 5 2022.  So that is much better.

gr5, 2023-07-02:
When it's time to make the installer, after you push your changes, the installer now gets built automatically.  Although the version should be updated in one of the files that creates the installer.

githubdoe, 2023-07-02:
Ok my drive could not be recovered.   My last backup was release 5.1  so my local github is very out of date.    Yet when I did a git status it said my git archive was up to date with origin\master.   

Does that mean I have the wrong branch and need to switch to branch xxx?  Is that part of the old master got renamed.  What do I do next?

atsju, 2023-07-02:
> When it's time to make the installer, after you push your changes, the installer now gets built automatically. Although the version should be updated in one of the files that creates the installer.

I'm trying hard to finish automated release. Please wait before doing a release. If I manage to do it, only thing you need to do is push a tag (e.g. v1.2.3) and you will get a draft release in GH interface with the installer uploaded and a release note with the changes. You can then update the text in the release note as you wish, test the installer one last time and just click push.

> Ok my drive could not be recovered. My last backup was release 5.1 so my local github is very out of date. Yet when I did a git status it said my git archive was up to date with origin\master.

> Does that mean I have the wrong branch and need to switch to branch xxx? Is that part of the old master got renamed. What do I do next?

Git status has not the up to date remote informations. You first need to do a `git fetch` this will update your remote informations. Then you can `git status` that will compare the up to date remote information with the local.
I'm available for next 12h. If you want I can setup a teams call with screen sharing any time to help you.

githubdoe, 2023-07-02:
Ah yes thanks.

githubdoe, 2023-07-02:
Now I don't have the right stuff for OpenCV and probably other libraries.  What was the strategy you guys decided on to setup those in order to build on windows for the includes and the libs of .dlls.

atsju, 2023-07-02:
Personally I don't build locally anymore. I let github do it for me, it takes 8-12 minutes each time.

If you need to build locally because of debug, you can follow steps in `build-windows.yml` or I can do it with you on teams or I shared my latest notes on how to windows build on groups.io

atsju, 2023-07-02:
OR if you are on windows 10, consider using WSL https://learn.microsoft.com/en-us/windows/wsl/install
This way you can "build on linux" in windows. And this is MUCH easier. But I never tried the debug part. Only build.

githubdoe, 2023-07-02:
Sorry I went to bed.  It is a must that I build locally with using Qt Creator as the IDE and it's links to the compiler and the debugger.  It would never work to take minutes to build between debug sessions.  I will do incremental builds and debugging hundreds of times while working on a feature.

atsju, 2023-07-02:
best is probably to follow [groups.io](https://groups.io/g/Interferometry/message/32887)
It's what is nearest to your previous workflow.

gr5, 2023-07-02:
I messaged dale directly but not sure if he got my email.  Anyway...

I am actually coincidentally building dftfringe right now on a fresh computer.  A laptop.  I'm doing the install.  You can definitely step through the code line by line.  I've done it.  But you can't step through other libraries like opencv if you do it my way.

So I don't build opencv nor armadillo nor qwt.  I just install those things to get the .h files only.  Only .h files!  I put the .h files where the .pro says to put them (the dale version of the .pro file).

Then I make sure to put all the dlls from a recent DFTFringe install in some folder somewhere.  I forget where I put them but I have notes.  So here are my notes...

gr5, 2023-07-02:
These notes are a mess but start at the top and skip down for more details if the notes near the top don't make sense.  Hopefully you only need the first 35 lines or so:
=====================================================
How to build DFTFringe (by gr@gr5.org)

================
Starting from scratch on laptop windows 11
I'm going to download the latest qt creator as qt libraries are not mentioned in the DFTFringe.pro file so I
can use any one I want.  Dale uses an older creator with QT 5.X (can tell by the name of the dll files)  but I just grabbed  qt creator version x.x.x with QT 6.4.1

On the qt.io website make sure you get the open source download.  There are at least 3 overall options: paid, "try it", open source.  Get the open source one.

After launching installer, I chose default locations.

I created a d: drive linked to a folder on c drive - network share the folder to "everyone" and
it will potentially prompt you to enable network file sharing and then do "share network drive"

followed instructions below to install qwt 6.1.5

Okay then I realized I have several modules missing:
charts multimedia multimediawidgets datavisualization

go to c:\qt\ find and open the MaintenanceTool.exe
select components:
qt -> qt 6.1.5 (or whatever is already installed) -> additional libraries -> qt charts, qt datavisualization, qt multimedia

I grabbed armadillo 11.4.4 - Dale used version 9.200.6 - it doesn't matter
https://arma.sourceforge.net/download.html
in wsl:
tar -xf armadillo-11.4.4.tar.xz
Create a folder d:\armadillo\
put extracted files in there, rename such that this exists:
d:\armadillo\armadillo-9.200.6\include\

================
some git notes
if you have unchecked changes and realize you want to put them on a new branch...
git checkout -b gr_arbitrary_wavefront

git checkout without the "-f" will never delete or overwrite your uncommited changes.  The above command creates a new branch based on the branch you were just in and now you can stage (add) and then commit your changes into the new branch.

================
notes "against" the pdf document
cmake 3.25.0
put into c:\cmake\

codeblocks 20.03 - got defaults which was not a mingw version but *did* include a compliler
put in default "program files" folder with the space in the path
'

I already have qt5.3.2 installed

Unfortunately I forget what I did.  I concluded 5.3.2 was not new enough.  So I think I went to uninstall it but then it made
me log in using my qt username/password but then it gave me lots of choices including installing specific versions.
anyway somehow I ran the web installer (or something like that) and ended up with qt creator 9.0.0 "based on qt 6.4.1"

Somehow I got qt 5.15.2 64bit with mingw.  I think the png helps explain it?

I had trouble with qt5.3.2 not recognizing a new module that is part of qt called "datavisualization" so I installed 

SEE w:\gr5web\qt_install.png

================

grab Dale's source code from github: https://github.com/githubdoe/DFTFringe

Dale builds on windows so I did the same.  I did this on windows 10.

Take an existing installation of DFTFringe and find all the dll files and copy certain ones to an otherwise empty folder.  These files.
If you copy too many you will get link errors when you compile the project:

[... 101 lines cut ...]
MSVC can do it.  Resource Hacker can do it.

The resulting file will be I believe in a subdirectory of your "home" folder.  For me that was here:
C:\Users\gr770\build-DFTFringe-Desktop_Qt_5_3_MinGW_32bit-Release\release

IGNORE NOTES FROM HERE DOWN

no: So you need qtcreator and also qt so I got both at once.  I went to qt.io/download and on lower right there is an
no: open source download thing - do that "go open source".  then I think I did the online installer?  Or maybe qt-unified something?
no: Somehow I installed qt 5.9.0 and it took over an hour to download (huge!) and it auto downloaded qtcreator version 4.14.0.
no: Even though I had just downloaded an older version.  Anyway I also had to create a qt account with password at this point.

no: start qtcreator.  Open the project file.  Select a build kit.  So do "tools" "options" and there are some tabs - the
no: key ones are "qt versions" and "kits".  For Kit I chose QT 5.9.0 Mingw 32 bit.  Somewhere I told it to create the Release
no: version of DFTFringe and not the Debug version (the debug version needs qwtd.dll).  Here's how to set to release:
no: click on the projects button left side with green wrench.  At the top - "edit build configuration" change to "Release".
no: While in there make sure the only thing that will build and run (list on left side) is the "desktop qt 5.9.0 mingw 32 bit".
no: Hit esc key to get out (I can't find close button).

no: Get qtcreator 4.2.1.  Google "downloading qtcreator 4.2.1".  Only trust things on *.qt.io domains.

cd /d C:\CMake\bin\opencv-build\modules\core && 
C:\Qt\Tools\mingw810_64\bin\gcc.exe -DCVAPI_EXPORTS -DOPENCV_ALLOCATOR_STATS_COUNTER_TYPE="long long" -D_USE_MATH_DEFINES -D_WIN32_WINNT=0x0601 -D__OPENCV_BUILD=1 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -DCV_CPU_COMPILE_SSE4_1=1 -DCV_CPU_COMPILE_SSSE3=1 -DCV_CPU_DISPATCH_MODE=SSE4_1 @CMakeFiles/opencv_core.dir/includes_CXX.rsp -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winit-se [... cut ...]  -Wno-strict-overflow -fdiagnostics-show-option -Wno-long-long -fomit-frame-pointer -ffunction-sections -fdata-sections  -msse -msse2 -msse3 -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG  -DNDEBUG   -Wl,--gc-sections -shared -o ..\..\bin\libopencv_core3412.dll -Wl,--out-implib,..\..\lib\libopencv_core3412.dll.a -Wl,--major-image-version,3,--minor-image-version,4 -Wl,--whole-archive CMakeFiles\opencv_core.dir/objects.a -Wl,--no-whole-archive @CMakeFiles\opencv_core.dir\linkLibs.rsp

githubdoe, 2023-07-02:
I have got builds working once again.  I had backups of all the external library stuff so that was easy.  I had forgotten about DFTFring_Dale.pro.  Once remembered and used then I can build.  

Sorry I have not yet read the comments just above this.  Been too busy with company and life matters.

githubdoe, 2023-07-02:
Thanks for the help guys.  I have been very busy as I said before.  I think I'm ok now.  I don't ever debug into the libraries either.  Sometimes that would be helpful but mostly not.   Now the only issue would be when/if I add files to the build both .pro files need to be updated.  But I can deal with that.  It present I abandoned the feature to fix wrap errors using diameter profiles.  It was fun to try and play with however.  So I don't have any new files to add now.

githubdoe, 2023-07-03:
The problem with the edge mask not being used was because the constructor of the mirror dialog read the enabled state of the edge mask check box too late in the process. Which was after it read the reduction value.    So I moved it up earlier where it now works.

atsju, 2023-07-03:
> Excellent. Do you want to go ahead and push your fix? Or even better, create a pull request with the fix? You can create a commit with only the fix in master and push it or do the full pull request process. Please don't accidentally commit your DFTFringe.pro.

If it's not too difficult, I would advise to use the PR process. This will ensure nothing is missing or brocken in the repository thanks to continuous integration.

```
git checkout -b DOE/myBranchName
git add theFilesYouWantToShareModifications
git commit -m "your message"
git push -u origin DOE/myBranchName
```

Then open the PR

For additional modifications
```
git add theFilesYouWantToShareModifications
git commit -m "your message"
git push
```

githubdoe, 2023-07-08:
I will work on saving the edge mask value in the outline file.
In addition I notice the internal edge mask width is not recalculated when the mirror outline changes size and it should be.  Since then the pixels per mm would be different.

githubdoe, 2023-07-08:
Let lump this all together so wait to merge.

githubdoe, 2023-07-10:
Here is a design decision to think about.  The edge mask should reside in the outline file if specified. Then when it is read back in if that igram is reprocess some time later it should be used.  Right now it is not stored in the .oln file.

But what to do with the value in the mirror config.  Should it be changed to match the value in the outline file?  That might surprise the user to discover their edge mask value in the config has been changed.  However maybe it should be.  Maybe if different than the outline file then a warning should be issued and ask if it can be changed.  If the answer is no then what?
1. Still use the .oln value?
2. Ignore the .oln value and use the default in the config.

githubdoe, 2023-07-10:
If the user answers no then we must use the config value because we can not change the config value.  It is only the config value that is used to calculate the edge mask later on in other code.  Too complicated to make a temp value to be used in it's place.

gr5, 2023-07-10:
Most people probably will never use this feature or they will use same value for a given mirror so I'm not to worried what you choose however...

I like storing it in the OLN file.  If same value there is nothing to do.  If OLN has zero silently use config.  Remaining cases ask user but make it clear that clicking "okay" copies OLN value to config.  Explain clearly that "mirror config edge value will change from Xmm to Ymm" or something similar.  Many of your other warnings are clear that the values don't match but unclear what happens when I hit the "yes" button.

githubdoe, 2023-07-12:
I have updated the code to save and restore edge mask info in the outline and use it when prompted by the user if it is different from that in the mirror config.

I did a commit and then a push.  I hope that is what I should do.  Now ready I guess for another review by any who care.

atsju, 2023-07-12:
Hi @githubdoe You changes do not appear on remote. But the process you describe is correct.
Was there any warning on commit or push ? Are you on correct branch ? Or more probably, **did you add the changes before commit ?** (or commit -a) 
Maybe you can do `gitk --all` this will show you a nice graph to understand what is on commited or not, pushed or not.

githubdoe, 2023-07-12:
Ok thanks.  Forgot the m after the -a when including text.  The git bash tool I use has tiny text for my poor old eyes and I did not notice the fail to begin with.

atsju, 2023-07-12:
Now it's there https://github.com/githubdoe/DFTFringe/pull/39/files/6af2a829248de88c0804934f4f0b6ee88853c77f..b0268c2acead681f9a0fb9cb127549a5b6407653
👍

gr5, 2023-07-12:
Excellent.

I have some very high priority outdoor tasks for today so give me about 24 hours.  If it rains today then I'll look at it then.  Most likely I'll look at it after dark tonight.

atsju, 2023-07-21:
I think this issue has been fixed as part of #39

gr5, 2023-07-21:
Yes.  Fixed in #39 and will be in release 6.3.0

closed 2023-08-03 by gr5