githubdoe/DFTFringe #41: Linting

Interferometry, 2023-07-08:
### Is your feature request related to a problem? Please describe.
There has been lots of great work on CI side recently.  Now that we have available builds for three platforms, it might be good to include some basic 'linting' on commit.

This could initially be used to solve problems with:
- various line endings
- tabs vs spaces
- includes

### Describe the solution you'd like
I'm thinking really basic to begin with. Line endings, duplicate includes, etc...

We can gradually introduce other rules over time, even to being able to enforce some code style

### Describe alternatives you've considered
Er... 
Don't do anything?
Or maybe there is some repo or editor setting that could achieve this?

### Additional context
I've have no idea how to do this, but I know it can be done in a fairly off-the-shelf manner from other projects I've been on. @atsju is this something you have

atsju, 2023-07-08:
I have never done it on server side. I have used hooks with uncrustify on client side. 

While I fully agree with you, I think we need to take this step by step and it's not a high priority now. I would push on doxygen format comments more than line ending and spaces. Feel free to propose something if it needs no specific work for historic developers. 

Note that we build only on Linux and windows. Mac is in progress.

Interferometry, 2023-07-08:
Yes, as I mentioned, it is something to introduce gradually. I know some of it, at least, can be done server side.

This is primarily of benefit to developers, not the CI.

At the most basic level, we want to avoid a sitution where every line in every altered code file gets highlighted as changed. It makes it harder to see what has actually changed.

Interferometry, 2023-07-08:
Okay so the line endings can be fixed with a repo .gitattributes file. Here is the example GitHub provide.

I'll include an appropriate one in my PR for the macOs version. Not sure why this isn't already set on my repo!

```
# Set the default behavior, in case people don't have core.autocrlf set.
* text=auto

# Explicitly declare text files you want to always be normalized and converted
# to native line endings on checkout.
*.c text
*.h text

# Declare files that will always have CRLF line endings on checkout.
*.sln text eol=crlf

# Denote all files that are truly binary and should not be modified.
*.png binary
*.jpg binary
```

atsju, 2023-07-08:
Ok your example will at least fix CRLF problems we already dealt with in the past. This would definitely help us.   
It won't remove white spaces on end of line for example or fix indentation. 

Feel free to push a PR and I will try it out to confirm it works as expected.

atsju, 2023-07-08:
Ah and maybe separate things. MacOS build should be a different PR than fixing line endings. 

1 feature == 1 PR

That's theory at least

gr5, 2023-07-16:
For some reason this line of code, found in several .cpp files, keeps getting flagged as "changed" in git diff even though it is not being changed.  The issue is related to crlf versus lf but I could not fix the issue and gave up.  Maybe this .gitattributes file will take care of it.  Anyway, here is the line of code:

gr5, 2023-07-16:
For example the above shows up 3 times in this most recent pull request: https://github.com/githubdoe/DFTFringe/pull/44/files

Anyway, fingers crossed.  :)

Interferometry, 2023-07-16:
My problem, it turns out, was caused by not having things configured in the main DFTFringe repo, but having different global configs on the various devices I use to connect. 

.gitattributes is the way to go.  I'll close this now.

closed 2023-07-16 by Interferometry