githubdoe/DFTFringe #43: [BUG] tools -> DebugStuff has no effect ?

atsju, 2023-07-15:
**Describe the bug**
tools -> DebugStuff opens no window and does nothing apparently.
Please confirm 

**Desktop (please complete the following information):**
 - OS: win10
 - DFTFringe version: I tried both 9be9dd and 57f007 from action installer and official 6.2 release
 - Build:  from GitHub Actions artifacts

** Note **
help -> version history is also affected but this is just RevisionHistory.html file missing in installer. I will provide a fix.

gr5, 2023-08-03:
I confirm that it doesn't do much.  I think the idea is that Dale is able to add something here some time in the future:

https://github.com/githubdoe/DFTFringe/blob/b89f2c8440f55ac3b783e693acf01fc57cf6cfb7/mainwindow.cpp#L1682C1-L1686C2

githubdoe, 2023-08-03:
Yes, there are times when I need something to be triggered on a menu command for debugging purposes.  That is the routine to do that when the associated menu item is clicked.

closed 2025-05-16 by atsju