githubdoe/DFTFringe #48: Possible bug

gr5, 2023-07-19:
**Describe the bug**
This line of code in zernikeprocess.cpp is probably wrong:
`            if (mask.at<bool>(x,y))`
All the other times the at() function is used it's y,x.  not x,y.  This probably only affects regions as outlines are symmetrical across the y=x symmetrical axis.  Regions are not.  So no one would notice the bug unless they use regions feature.

I'm not certain but I think this code is used in the zernike smoothing feature which is relatively new and probably not used by very many people.

atsju, 2023-07-19:
permalink : https://github.com/githubdoe/DFTFringe/blob/1daf63a5dba50d9512b41a11263c122a84df85af/zernikeprocess.cpp#L1084

gr5, 2023-08-13:
Dale can you click on the link above and confirm that it should by "y,x" and not "x,y"?  This appears to be in your "new" zernike smoothing feature.  Maybe it doesn't matter?  Because the mask is circular and perfectly centered in the matrix?  Or maybe this messes up some pixels on the edge of the outline?

githubdoe, 2023-08-13:
That sounds like a reasonable assumption.  CV:Mat are accessed by row order (y) first.  By habit I tend to write (x,y) however.

gr5, 2023-08-13:
Should I just swap it then?  I can try to test but I'm not sure how to prove it's currently wrong.  After swapping I would of course test the feature and I can add a large region but I don't think I'll see any difference as the "region" is 47 zernike smoothed and the rest of the wavefront would be something like 300 zernike smoothed.

Also I think the region smoothing removes the regions from the wavefront after the smoothing is complete?  I'm not sure of that detail.  And if that's true then regions won't affect this bug and only the outline.

I vote to just swap it, do some limited testing.  What do you think Dale?

githubdoe, 2023-08-13:
I think it is a bug.  However it does not matter since the mask is symmetric.   So I think you will not see any difference unless you can trick it with a region but like you say.

closed 2024-08-02 by gr5