githubdoe/DFTFringe #49: DFTFringe Next Version Label.

githubdoe, 2023-07-20:
I'm leaning to calling it Version 7.0  Because from a users perspective it now supports Mac and Linux users.

That way we can refer them to Version 7 when they what to know which version no gives them support to start with.

atsju, 2023-07-20:
Well build is supported on linux. (not on mac yet). But we do not provide an executable for linux (yet).

I'm OK with both v7.0.0 and v6.3.0

githubdoe, 2023-07-20:
Ok, So in that case
I don't care either.  I don't know a lot about providing Linux builds but I thought that is hard and that no one build will work for all.

Interferometry, 2023-07-21:
I'll endeavour to get the macOS build updated to this version over the weekend too!

Interferometry, 2023-07-25:
We have a macOS version 6.3.1 now also!
[DFTFringe-6.3.1-macOS-Universal.zip](https://github.com/Interferometry/DFTFringe-macOS/blob/main/download/DFTFringe-6.3.1-macOS-Universal.zip)

Other macOS releases can be found here:
https://github.com/Interferometry/DFTFringe-macOS

atsju, 2023-07-25:
> We have a macOS version 6.3.1 now also! [DFTFringe-6.3.1-macOS-Universal.zip](https://github.com/Interferometry/DFTFringe-macOS/blob/main/download/DFTFringe-6.3.1-macOS-Universal.zip)
> 
> Other macOS releases can be found here: https://github.com/Interferometry/DFTFringe-macOS

Good news.

Is there any reason to not merge the macOS version modifications here ? From maintenance point of view this would really make sense.

And second, there are good chances you are breaking some of the open source licences by not making your modified sources available.

gr5, 2023-08-05:
I want to go to v7.0.0 on next build because the format for the oln files has changed and it will be easier to explain to people who bring it up that it changed in v7.  Also Julien already put some code comments marking which code reads V6 oln files and which read v7 oln files so if we don't go to v7 we also need to update the code comments, lol.

gr5, 2023-08-05:
Actually they aren't comments - they are actually built into the name of the functions that read and write oln files. :smile:

gr5, 2023-08-17:
The v6 version of oln files are bad.  They can even crash DFTF (I strongly suspect).  The v7 version will reliably contain the information from the v6 files that is retrievable.  The oln files are written over when you process the igram (both v6 software and v7).   While playing with the outline, before you hit "crop" or "done" you can still re-read the oln file.

If you have an inner obstruction outline (for perforated mirrors), I believe the odds are 25% that the file has an ascii 26 (CTRL/Z) which terminates reading of the file so the rest won't be read.  If only the outer circle then I think the odds are around 14%.  There are other errors related to ascii 10 that can cause errors in another 3% of the files.  The statistics are all from memory but I posted the statistics somewhere probably in a github comment or an email to dale.

Anyway I don't see why we should save the original v6 oln files.  The current DFTF writes over them also when processing the igram.

atsju, 2023-08-17:
OK that's enough justification for me. 
Then maybe we can release v7.0.0 in coming days once the Dale finished review of memory leaks ?

I'm looking at logging and crash dump right now. Maybe for v7.1.0 ...

atsju, 2023-08-17:
> Before we release 7.0.0 I will do lots of testing of the actual code (e.g. analyze igrams, try out the more popular features like ronchi, astig removal, auto outline, and a dozen or so other features.

Given small amount of users, it's probably best approach but I'm sorry you get so much to test manually.

closed 2023-09-14 by gr5