githubdoe/DFTFringe #50: Each recalculate grows all the regions

gr5, 2023-07-21:
**Describe the bug**
regions grow on every recalculation.  This is bad.  Someone might want to change the zernike checkboxes (e.g. remove astig) then you have to hit recalculate and now the regions grow again.

**To Reproduce**
Steps to reproduce the behavior:
1) analyze an igram as usual but create a small region.  
2) When final analysis is done remove gaussian filter and the region is easy to see in the wavefront.
3) Click recalculate and the region grows by 2 pixels or 10% (whichever is smaller).  Sometimes you also get unwrap errors as the regions keep growing

**Expected behavior**
Regions shouldn't keep growing.

**Fix**
Well I'm not sure we should be growing them in the first place but one could add a flag to the region to keep track of whether it has been grown or not and never run the expansion code the second time.

gr5, 2023-09-14:
Fixed in PR#57

closed 2023-09-14 by gr5

atsju, 2023-09-15:
#57