githubdoe/DFTFringe #74: astigstatsdlg display issue

atsju, 2023-08-14:
This is what astigstatsdlg should look like after QTcreator
[image]

This is what it actually looks like (before you ask, picture is from 6.2, so before I "messed" with ui files)
[image]

This is why I did not see the missing `i` effect in #71 during my tests. I couldn't use/test the affected code.
I also found out a different place in original code where `i` was not incremented and should be. I will try to fix all this...

atsju, 2023-08-14:
OK for this feature to work, I need to make "groups" of pictures. So this is NOT a bug.
It has been difficult for me to active this "group" feature. I will do some more tests as I found some strange things while investigating. I will reopen if I confirm a problem

atsju, 2023-08-14:
@githubdoe I have not been able to create "groups" other than renaming manually in DFTFringe. While reading the documentation I expected that folder name is taken into account. How do I do that ? is it normal.

Second question: SD are not same in both windows. Is this normal ? 
[image]

And third I will do a PR because I had really hard time reading the code and understanding if an `i++` was missing. Spoiler, it was not, but code was written as if it was.

atsju, 2023-08-14:
and 4th question, is the black circle well located in my picture ? Seems strange to me but maybe it's normal.

atsju, 2023-08-14:
For second question about SD not being the same I confirm they are not computed the same way. The one in distribution window is computed with cv::meanStdDevand is wrong

githubdoe, 2023-08-14:
I think the best fit around the mean but the mean does not look like it is in the right place.

Perhaps comparing with an older version of DFTFringe.  Maybe even back to the rev 5 series.   The groups should be grouped by directory but of course we don't want to go all the way back to the root of the project.  Instead I did something I don't remember but tried to figure out the root of the rotations and then use the directory name just to the next dir of that.

The STD difference may be because I'm computing a different group of things and is intentional.  Maybe where one is the STD of points inside a group and the other is the STD off all the means of groups.

atsju, 2023-08-14:
Ok I will try much older versions for groups. I already tested 6.2 and it is same.

I was working on stddev, and I have been able to check with same data. I confirm it's openCV that lacks precision. 
I will open issue on OpenCV projet and see how we fix in ours if it's misuse or bug.

githubdoe, 2023-08-14:
This is of course hard to create data sets to test this.  What you have to do is create a surface with some astig perhaps xastig of .1
Then rotate that surface every 45deg.  Then to each of those surfaces add a constant stand astig perhaps x of .05.  Adding astig to each can be done manually with the simulate dialog by importing each and then manually adding to each.  

Then of course you need a slightly random variation of the mirror astig to simulate air current variations.  Then sort all of them out into groups of directories.  There may also be a way to just rename them with "/" separators.  Using the right mouse button on the wave front names in the wave front list.

I cheat.  I use a real data set from a well analyzed mirror with many rotations.

This dialog is very complex but shows a simple idea.  That when the mirror has astig the Cartesian plot of the astig  with many rotations will be a circle around the mirror stand astig.  The best fit is trying to draw that circle.  It then gives you stats like STD to help you understand if you have enough samples to make a reasonable guess.   How you best fit points to that circle is the problem since some people only do 2 rotations.

githubdoe, 2023-08-14:
After looking at the code.  The group name is usually the last folder in the path to the file.

atsju, 2023-08-15:
About Standard deviation. One is using standard deviation and the other (`cv::`) is using "sample standard deviation". See https://en.wikipedia.org/wiki/Standard_deviation and https://github.com/opencv/opencv/issues/24155#issuecomment-1678216348. It might be what we actually want. Should we update documentation instead of code ?

> I wonder if the Y astig mean is computed correctly for that black circle.

[image]
I read your documentation and it seems to be correct. Black circle is not mean of groups but rather fitting the circle groups would draw if we had an infinite number of rotation groups. Looks correct now that I understand what it's supposed to do. 

> After looking at the code. The group name is usually the last folder in the path to the file.

Yes this is what I deduced from looking at the code also. But while using DFTFringe it seems that the unique group name appearing on group mean value is always the name of last opened folder. I cannot add data from multiple folders in one step and adding data from a different folder will rename the group to last folder for all data. I did not find an other way to do proper groups than right click + rename to group/...
Re-reading the code QSetting lastPath is used always so it cannot work automatically. We might load pictures with 2 last parts of name instead one like `folder/PIC123456` instead `PIC123456`.

githubdoe, 2023-08-15:
Here is what I get when loading some of my data for the RR mirror.  I don't know if this helps but it is what I expect.  The black circle did a good fit to the 4 data points.  The names are as I would expect as well.     The most desired position of the center would be the mean however fitting a circle with three points will not get it there.  Usually there should be an even number of rotations since we want pairs.  Then I think the center becomes closer to the mean.  But maybe not always.    Anyway your first posted graph of the black circle does not look right.   Try it with an older version to see if it is different.  If not then.....I don't know.[
[image]
](url)

atsju, 2023-09-12:
I close as initial issue was just a bad usage. I did not have any further issue with this

closed 2023-09-12 by atsju