githubdoe/DFTFringe #79: regions near edge of mirror can cause dftf to crash

gr5, 2023-08-22:
**Describe the bug**
The code that expands the region size by "10%" can expand it beyond the matrix and cause a crash or sometimes create an empty wavefront.  Different behaviors depending on unknown factors.

**To Reproduce**
Create a region well above and left of the mirror before hitting the crop button.  After cropping the region is in the negative x and y space.

Process the igram normallly and program crashes.

I already have a fix.  Will create pull request asap.  The problem is that I keep the region points within the matrix but I need to repeat that code after the "10% expansion" occurs.  Very simple fix.  If all the points for example are above and left they all end up at 0,0 and the expansion is infinite.

Even if all the points are within the igram, after cropping, if the region is big enough and near (or a little beyond) the edge of the igram, 10% could possibly push the region just outside the matrix.

**Screenshots**
[image]

atsju, 2023-08-23:
You can close this one I think. 
> Fixes issue #79

does not work. I needs to be "fixes #79" or other [compliant wording](https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue#linking-a-pull-request-to-an-issue-using-a-keyword)

gr5, 2023-08-23:
fixed in PR80 which is planned to be in release 7.0.0.

closed 2023-08-23 by gr5