githubdoe/DFTFringe #82: many dialogs don't close when application closes

gr5, 2023-08-23:
**To Reproduce**
**zernike smoothing**

**For the following, first process an igram to get a wavefront.  Then:**

1. Do "tools" "smooth current wave front".  Before doing anything with this dialog close the app.  The dialog stays running and the DFTF icon is still visible as an actively running application.

1. tools  wavefront inspector
2. view show statistics of loaded wavefronts

**needs igram loaded but not a wavefront**
1.tools outline helper

**others - just launch dialog and close window:**
1. tools user color map
1. tools test stand astig removal
3. tools camera calibration
4. tools astig stats
5. config mirror
6. config preferences

tools tilt versus astig analysis - first you need to create a zernike csv file.  With one or more wavefronts loaded do "file" "save zernike values in csv".  Then run the astig analysis off the csv file.

**Expected behavior**
all dialogs should close when closing DFTF main window.

**Desktop (please complete the following information):**
win11, master branch as of 8/23/23 - probably will become v7.0.0

atsju, 2023-08-23:
This one is for me I think. I will love to write all these clean destructors :)

atsju, 2023-09-23:
### wft needed

- [ ] 1. Do "tools" "smooth current wave front". Before doing anything with this dialog close the app. The dialog stays running and the DFTF icon is still visible as an actively running application.
- [ ] 2. tools wavefront inspector
- [ ] 3. view show statistics of loaded wavefronts
- [ ] 4. Click full screen above the 3d window, alt-tab to app, close app
- [ ] 5.right click upper right wavefront coutour chart. Choose "full screen". alt-tab to app and close app.

### needs igram loaded but not a wavefront
- [ ] 1.tools outline helper

### others - just launch dialog and close window:

- [ ] 1. tools user color map
- [ ] 2. tools test stand astig removal
- [ ] 3. tools camera calibration
- [ ] 4. tools astig stats
- [ ] 5. config mirror
- [ ] 6. config preferences