githubdoe/DFTFringe #85: broken "show only average" in astig stats

atsju, 2023-08-27:
**Describe the bug**
"show only average" does still show all points.
[image]
probably introduced by my change in #75 :( 

**Desktop (please complete the following information):**
 - OS: Win
 - DFTFringe version: v7.0.0
 - Build: official

closed 2023-09-05 by gr5

githubdoe, 2023-10-12:
Yes I think it is.  For some reason I don't think I'm getting all emails on the issues.  For instance I did not get my posting of this issue but maybe github is smart enough to not email the originator.  So it is possible I just lost the first posting of this issue.

gr5, 2023-10-12:
I wasn't getting notices of new pull requests for the last year; so a week ago I finally googled how to get all notifications and I just did that.  Some setting on github.

PR #86 (2023-09-05): [bug fix] "show only average" in astig stats wasn't removing display
I'm getting better at QWT and QT. Thank chatGPT on this one also.

fix #85 by using QWTplotPicker to display what we need to. I have been able to fix the bug and still keep the feature.
Moving mouse +/-5 pixels around the data will display the wavefront name
commit b7d3cca1 (2023-09-05): fix #85 display wavefront infos using plotPicker instead other means