githubdoe/DFTFringe #87: [BUG] wavefrontList is bugged and could be improved

atsju, 2023-09-09:
**Describe the bug**
Double click on a wavefront does no permit to rename it.
Once you renamed one wvft with right click they become editable/renamable with double click.

**Improvement**
I would like to be able to rename using F2 which is standard renaming shortcut on windows
Up/down + enter should select/display the wavefront. Today a click is mandatory and this bothers me as a user.

I'm already working on the code so this issue is more about getting a feedback while working on it.

**My basic proposal (this is all nativelly supported by QListWidget)**
Add F2 to rename
Add up/down + enter as a way to display a wavefront

**Potentially controversial proposal**
- Right click is for context menu. I propose that we remove the right click as renaming. Replaced by both F2 and double click. It's more natural even if you used right click for years.
- I have seen in the code that simple click and double click on waveform have different effect. I did not dig enough to make clear recommendation about it right now but I think all should be managed by simple click and remove double click effect.

Both the double click and right click are a bit of masked features that a user will not naturally find if he doesn't know about it.

atsju, 2023-09-09:
Slots and signals between simple click and double click are such a spaghetti.
-----
simple click
waveFrontClickedSlot
sendSurface
currentNdxChanged
currentNdxChanged

---------
simple click    => double click implies a simple click
waveFrontClickedSlot
sendSurface
currentNdxChanged
currentNdxChanged

double click
wavefrontDClicked
sendSurface
currentNdxChanged
simple click
waveFrontClickedSlot
sendSurface
currentNdxChanged
currentNdxChanged
---------
I will lean this up. 

Moreover, there is an additional bug. If you have several wft with same name or same ending (typically average have all same nmes) then the double click will display first one because of how `void SurfaceManager::wavefrontDClicked(const QString & name){` is implemented

atsju, 2023-09-09:
While reading the code I understand why wft names are greyed out text when displayed. But this is not very smooth. 
When loading a bath of igrams, all are black and then they grey out when clicking one after an other. I will try to improve this.

atsju, 2023-09-11:
one more issue:

select multiple wft in the list then right click to rename one. 
The name in metrics will nerver match the wft name again even on reload.

githubdoe, 2023-09-11:
I don't know if you knew that if you right click on a wave front that lets you change the name.  I don't mind adding other options however.  That name change is only for that list and will not change the name of the source file unless one selects that wave front to be saved in the file menu.

atsju, 2023-09-12:
> I don't know if you knew that if you right click on a wave front that lets you change the name. I don't mind adding other options however. That name change is only for that list and will not change the name of the source file unless one selects that wave front to be saved in the file menu.

Yes you mentioned it once and I use it often now. 
My problem was more about people not finding it because right click is not a rename shortcut in general.
It hink I'm achieving something very user friendly and solved all corner case bugs that I discovered along the path. I don't do more as before with file names, rename will still have same effect and not rename source files.

My main additions: 
- you can now select and navigate with arrow keys + enter.
- Rename can be accessed with right click, with F2 and with double click.

Detail of my changes are listed in the PR.

Do not hesitate to test the artifact build of PR #88 (or wait until I finish the PR to test all at once)

PR #88 (2023-09-17): [feature] Enter displays wavefront. Double click and F2 rename from the start
fix #87 

- [x] press enter displays wft
- [x] F2 renames wft
- [x] ~~right click does not rename anymore~~ Do a real right click menu instead as per George recommandation
- [x] double click has no special effect anymore
- [x] greying out wft names is strange behavior
- [x] create a shortcut for delete wft (dell key)
- [x] any edit will trigger correct signal to reload the wft and have name displayed correctly everywhere
- [x] right click menu can rotate and invert
- [x] fix display bug on delete

I think I achieved  [... cut]
closed 2023-09-17 by gr5