githubdoe/DFTFringe #94: read zernike values does not work

githubdoe, 2023-09-19:
After reading zernike values from a .zrn file the create wave front creates one with all zero zernike values instead of the ones just read.

gr5, 2023-09-19:
@atsju want to look at this?  I can't do much related to DFTF until probably Thursday night.

atsju, 2023-09-19:
I may have some time this weekend to look at it.

@githubdoe is this is v7.0.0 or specifically in latest master branch ?

atsju, 2023-09-19:
Ok I just see it's v7.0.0

Original report : https://groups.io/g/Interferometry/message/33768

atsju, 2023-09-22:
I have not been able to reproduce so I asked question to Sorin and got answer:

> it seems that I created a wavefront with all zernike terms being disabled.
> I did what you said: "disabled all" then "enabled all" and obviously worked.
> I simply didn't notice that all the imported terms were not checked. Sorry for inconvenience.
> So no need for the zernike list and the registry keys.

@gr5 Conclusion: there is not a bug. 
However, the user interface is not exactly user friendly.
- Button is labelled "disable all" when all terms are already disabled
- terms are not enabled by default

Maybe rename issue to "[feature request] improve edit zernike terms interface" or close this and create a new one.

githubdoe, 2023-09-22:
IIRC this dialog is used by a couple of different features each with a little different function but needing to display and edit the list of zernike values or perhaps one just to display them.