githubdoe/DFTFringe #97: Version 7 issues

githubdoe, 2023-10-12:
Playing around I discovered that if an igram does not have an outline and earlier I gave an outline also an ignore region so that it became the default outline (save in the registry).  Then that region will appear.  Then if I delete that region and process the igram the next igram without a region gets that deleted region.  I once found that I was not updating the region in the registry when I deleted the outline.  Is that still the case?

Greg.  Check it out please.

Issue 2.  
Astig stats display.  When display only averages is check just the average of each of the rotations should be shown.  I know longer remember exactly how I figured out what wavef ront were in what rotation.  Probably used the directory structure.  Or maybe just looked for the some text like avg in the wave front name.  Anyway all wave fronts were plotted instead when I expected only the average to be plotted.

gr5, 2023-10-12:
I can look at these.  First one seems pretty simple.  Probably there is no code written to delete form registry - easy enough to fix.  The next 6 days I'm less busy than usual so I should have time in the near future.

githubdoe, 2023-10-12:
Now it seems that the region issue does not always happen.  That makes it hard to find.  Hmm.

githubdoe, 2023-10-12:
I looked at the code for the average issue and now remember that it computes the mean for each group and thus does not need to have a wave front with the name of average in the list of wave fronts.  I think.  Why it does not work I don't know but perhaps I wacked it at some point.

githubdoe, 2023-10-12:
The averaging issue works when I compile my version of the code which I think has not been updated with the version 7 source.  So it broke under version 7.  There is a place in my code where it checks to see if the average only check box is checked.  If not then it plots the other points.    

I'm scared to update my code to the version 7 source.  I don't want to mess up my last good build (but older version).    I know that puts me out of date but I want something to hand on to that I know works for me.

githubdoe, 2023-10-12:
Yes I see that astigstatsDlg plot routine deleted important code and is the cause of the issue.  I will let you guys fix it.  Look at where the average only check box is checked for the difference.

gr5, 2023-10-12:
I'm curious what commit you are on.  

From QT Creator do menu "tools" "git" "local repository" "log" and the first line that is shown will have what I need.

atsju, 2023-10-13:
I kindly recommend that you put meaningful title to issues (you can edit) and separate different problems in different github issues.

`Version 7 issues` with a list of 2 issues does not help tracking.

May I recommend this issue becomes something in the line of "region not deleted correctly between pictures"

gr5, 2023-10-13:
I just confirmed that the averaging checkbox works fine for the latest build which I installed from bottom of this page (it doesn't look like a link but you can click on it):
https://github.com/githubdoe/DFTFringe/actions/runs/6441876548

gr5, 2023-10-13:
So issue 2 is fine.  I'll try to duplicate issue 1

gr5, 2023-10-13:
Issue 1 is a new bug.  So I played with the program and I couldn't get it to ever update the registry values for "lastRegions".  I looked at the code and I realize that the old "writeOutlines" function (now called writeOutlinesOldV6() but never executed) saves to lastRegions but the new writeOutlines() does not.  **I will fix this now.**

Work around for V7.0.0 is to select region radio button and click button that says "Delete Outline" (which is sort of hard to find).  This will clear out lastRegions in registry.

gr5, 2023-10-13:
Closing as both issues are fixed now in latest source code and both fixes will be in next release (probably 7.1).

closed 2023-10-13 by gr5