githubdoe/DFTFringe #98: Version 7 is very slow computing DFT

githubdoe, 2023-10-12:
Version 7 takes up to 7 seconds to compute the DFT of some of the igrams I have tested compared to fractions of a second for my older version.

The igrams I have just tested are only 640 x 640 big with the mirror diameter being only 160 x 160 pixels.  So that makes the DFT very small and should not take any time to compute.

I also see a pause after compute surface is pressed that can take several seconds on version 7 and not on the earlier versions.

githubdoe, 2023-10-12:
However after running my older version now version 7 is fast.  What?  Strange.

githubdoe, 2023-10-12:
Version 7 can be slow taking many seconds before it displays the DFT in Batch mode.  Version 6 never takes more than a fraction of a second of the same Batch of igrams.  Version 7 will process it fast if in manual mode.   Even in auto mode version 7 can pause for many seconds before continuing the batch.

githubdoe, 2023-10-13:
For my latest testing I find that it is fast always if I don't use the Batch mode.  However using the Batch mode then each igram takes a different amount of time from 1 second to 10's of seconds to present the DFT or surface yet all igrams are the same size.

So somehow perhaps the Batch mode looping is halted for some time for some reason.

gr5, 2023-10-14:
I tested this with version 7.0.0 and it seems fine.  I did batch mode manually and auto and it seemed about the same.  It was about 2-3 seconds for each processing whether I had it on 600X600 pixels or 160X160.  Did you have "Auto Outlines" enabled?  These were my settings.  Did you have any other checkboxes checked?  Just for an experiment please check and uncheck any boxes that were not checked to be sure the setting is off as I've seen bugs before where it looked like something was unchecked but the operation was performed anyway but after clicking the box twice it was fine.

[image]

gr5, 2023-10-14:
Also there is a "bug" where saving files on a network drive (especially wavefront files) can take many many seconds.  Could it be you have "save wavefront file" checked and it is saving them to a network drive?

githubdoe, 2023-10-14:
No I don't save files over the network.  Auto can be checked or unchecked.  Save file can be checked or unchecked.  It still happens with either of those settings.  Will investigate conditions when I have more time.

atsju, 2025-11-10:
Closing this. No one will investigate and it's too old now. 

Just open a new issue if problem is still present in later release

closed 2025-11-10 by atsju