githubdoe/DFTFringe #99: Possible issue with outlines

githubdoe, 2023-10-12:
Perhaps not a bug but a consequence of how outlines are now saved and it can  not be helped.

I tried to use  version 6 of DFT on the igrams I processed with Version 7 to do some comparisons.
I was trying to use the Batch feature in Auto mode where it would use the existing outline and process all the igrams without intervention.

It claimed there were no outlines for those igrams.  I suppose it is because they are now in the new format and can not be read by the old versions.  If so that is unfortunate but I suppose can not be helped.  It will hit those people like Ric who quickly revert to an old version after thinking 7 or later has bugs.  But I don't know any way to avoid it.

I was trying to see if the Batch feature was the cause of the slow computing of the DFT in Version 7.  I have found that version 7 intermittently has slow computes when I use it's Batch mode.  It may also have them in single mode.  However when I try it in single mode an igam that was slow in batch works fast.  Then when I go back into batch mode a different igram will become slow.  So confusing to figure out what is going on.  I was hoping to see if it happened in version 6.  But I will have to create outlines compatible with it first.  I don't know why Batch mode would be any different than single mode but the issue happened first last night when I was Batch process a set of igrams for Pierre Lemay.  (a new Bath and DFTFringe user.)

gr5, 2023-10-12:
Yes I've run into this many times especially before 7 was released and I needed to compare results from version 6 and 7.

My solution was to create a separate folder for the version 6 tests and the version 7 tests.  With identical files in both folders but the version 7 folder would be incompatible after it would convert the OLN files.

gr5, 2023-11-01:
Another solution would be to save in both formats every time (the old save code is still there).  "OLN" and "OLN2".  That way it could be backwards and forwards compatible.

atsju, 2023-11-01:
Automatically create a copy of old oln to avoid "surprise deletion" Of old format may be.    
But I wouldn't recommend keeping back compatibility. We know old format has flaws, and adding features will make maintenance more challenging than it already is. Unless more people than Dale, you and me really need this, that would be my recommandation. I don't see a normal usage flow where people still need v6.

closed 2023-11-10 by gr5