githubdoe/DFTFringe #101: Version 7 scan does not create a sequence of foucault and ronchi image if ROC multiplier is other than 1.

githubdoe, 2023-10-14:
One the Foucault Ronchi tab

Set start offset to -.1  set end offset to .1  set step size to .01
Set the ROC multiplier to 10

press scan it should change the roc offset from -.1 to .1 in steps of .01 displaying those images one at a time and then stop.

Instead it display only the very first offset image and thinks it done.

Then start and end computations use the wrong value.  I bet it mistakenly uses the ronchi roc offset.  Which was added more recently and modified by the ROC multiplier to get it to display better since ronchi needs larger offsets to display meaningful results.

If I set the Ronchi ROC multiplier to 1 it works as expected.  I think it would work if it watched the Foucault offset instead of the ROC offset.  As I said there use to be only one offset until I added the ronchi ROC multiplier.

Sorry guys.  Can one of your look at that?

gr5, 2023-10-14:
sure.

gr5, 2023-10-14:
It works fine for me.  It did 21 steps.  This is right after it finished.  I tried the version in "master" first and that worked so then I tried 7.0.0 and that worked also.  Maybe if you look at the settings shown in this screenshot below you can find the key difference?

[image]

githubdoe, 2023-10-14:
It failed for both me an Pierre yesterday.  However since then I have changed my settings and now it works.  So I have lost the exact config that caused it.  It must not be caused by my first guess.

gr5, 2025-05-05:
We can't seem to duplicate this so I'm closing this.  Feel free to reopen.

closed 2025-05-05 by gr5