githubdoe/DFTFringe #106: Version 7 and maybe older display "TextLabel" in Foucault view instead of number.

githubdoe, 2023-10-23:
[image]
The center between Ronchi and Foucault images should be a number scale.  It worked back in version 4.0 but maybe not in version 6 and later.  However this display shows version 7.0.  I had though it was my usual problem trying to use my 4K laptop screen but this image is from setting my screen down to 1920 x 1080.  Probably related to pixels per inch and windows and qt problems I have never understood.

That is text becomes too tiny when I use my 4K display in 4K.  So some places I changed the forms to force it bigger but not everywhere.  I gave up and hopped most users would use the smaller screen resolution.  But that also creates issues.   Oh, well.  Hope someone can figure out something.

gr5, 2023-10-24:
Sometimes I see that and then it suddenly goes away.  I think I only see it when there are no igrams processed perhaps?  I'll look at it this week.

As far as I know this tab usually looks fine.

gr5, 2023-10-24:
Maybe we fixed it since 7.0.0 and it's time for a new release?  Anyway I'll look at it.

atsju, 2023-10-24:
I'm experiencing what Dale describes but never reported it and then forgot about it.
I this this is not fixed already.

A new release sounds great. I have a bit less time for DFTF these days and it might last.

githubdoe, 2023-10-24:
Yes it then worked for me after not.  Hmmm.  To bad I don't know the difference.  I think somehow Qt decides it can't display the desired text in the field given so then displays "text label" instead.  I tried to track it down many versions ago and IIRC thought I had fixed it by changing object spacing in the window.

gr5, 2023-10-27:
The function that updates the labels is called: 

**draw_ROC_Scale()**

It is only called if you change the scale among mm/inches or if you check/uncheck "auto" next to step size.

It needs to be called when this whole page is initialized with the diagrams and perhaps when other things are called.  I'll play with it a bit.

gr5, 2023-10-29:
Dale - I figured it out - you broke it in ver 6.2.  Here is a link to the commit that broke the numbers to display in foucaultView:
https://github.com/githubdoe/DFTFringe/commit/af550d60c30b4f13e2ad3f03a6d4ec82c5fd2916

Notice at the top of those changes you can see in red that you cut some lines of code starting at line 103 and moved them from the function setSurface() to the function on_pushButton_clicked() (the "70%" button).  I think you meant to copy them and not to move those lines of code?

I think the fix is to put those lines of code back into setSurface() (and also keep them for when you click the 70% button)

Comments before I try that?

githubdoe, 2023-10-30:
thank you sounds reasonabl.

gr5, 2023-10-30:
Okay so I need you to do a code review when you get a chance.  My changes are purely putting the code back the way it was (in two spots, one for this issue and one another spot for issue 108).

githubdoe, 2023-10-30:
Sorry I'm confused.  I can't find any code to review associated with this.  I'm use to finding buttons up at the top the request that let me see the code and then respond to the review.

gr5, 2023-11-01:
Fixed by PR #109

closed 2023-11-01 by gr5