githubdoe/DFTFringe #108: Version 7.  does not display effects of central obstruction in most dispays.

githubdoe, 2023-10-24:
This may be a duplicate.  I just tried to start a report but hit return earlier and that then may have created a blank bug report.  Could not figure out how to remove it or change it.  If in fact it was recorded.

In the mirror config set a non-zero obstruction.

It should show up in all of the displays.  It only shows up on the profile plot.  But should show up on the 3d, Contour, Star test, PSF, and MTF.  I don't remember if it should also on the ronchi foucault.

It does work on version 6.4.

gr5, 2023-10-25:
It's quite confusing because you can generate the inner mask 2 ways:
1. With mirror configuration
2. By drawing it manually

I definitely changed some of the code here - it's a simple and small change but was important to get other things working like averaging and rotation when there is a central obstruction.

So just like DFTF fills in the regions with a kind of zernike smoothed data, it now also fills in the central obstruction with smoothed data.  It's almost the same code.  It's in the function  fillVoid().  Initially I just added filling in a few pixels within the border of the inner obstruction but then realized it's possible to move the "center" obstruction anywhere you want.  So the averaging code handles this rather than forcing people to place it in the exact center in the gui (which would have been a ton of work).   Anyway, now I just fill in the entire center (entire masked region).  

**Having said all that - I thought all the displays still showed a black circle in the center.**   I mean there is still a mask there.  It should still be a black circle in most of the displays.

HOWEVER, I'm wondering if you averaged a bunch of igrams together but the "first" or "selected" one didn't have any obstruction?  I'm thinking somehow while processing many igrams, or saving and loading igrams or something, the central obstruction information got lost.

The averaging function takes the inner and outer circle borders from the "first" chosen igram.  So if that one doesn't have an inner obstruction then the average one will also have none.

Or this could be a read/write OLN file issue.  Or this could be an issue of when the mask is created based on the mirror configuration.

I'll look at this later.  This week has been busier than expected so far.

githubdoe, 2023-10-25:
Don't get confused between the central obstruction and a center igram mask region.

The central obstruction is added to the analysis as the user alters the mirror dialog.
The central igram mask is added from the igram using the central outline.  It is necessary when the mirror has a hole in it and so the igram has no data in the central area.

The central obstruction of the telescope specified by the mirror dialog can be varied any time but is not controlled by the outline or the .oln file.

The central obstruction can be varied by the user independent of what the source wave front data has in it.  Both of course should create a hole in the mask.  Perhaps you lost the central obstruction mask hole creation.

githubdoe, 2023-10-25:
SurfaceManager.cpp lines 660 in makemask created the central obstruction into the mask in version 6.4

gr5, 2023-11-01:
Fixed by PR #109