githubdoe/DFTFringe #112: 7.0.0 crashes when doing wavefront smoothing

gr5, 2023-11-07:
**Describe the bug**
Menu item "tools" "smooth current wave front" - if you increase size of zernike terms or just click "create wave front" button, it crashes

**version**
tag 7.1.0 (not released) - the version built by github.  

When I build it myself in QT in debug or release mode it does not crash.  I have a different version of the armadillo .h files.

**stacktrace**
```
[2023-11-07 10:58:34.004] [critical] Unexpected issue. Stacktrace:

 2# 0x000000006FC56FA8 in C:\Users\gr770\DFTF710RC\libstdc++-6.dll
 3# 0x000000006FCFD093 in C:\Users\gr770\DFTF710RC\libstdc++-6.dll
 4# 0x000000006FD05E12 in C:\Users\gr770\DFTF710RC\libstdc++-6.dll
 5# 0x00000000005B9835 in C:\Users\gr770\DFTF710RC\DFTFringe.exe ...ingw32/8.1.0/include/c++/bits/basic_string.h:211
 6# 0x0000000000504FF4 in C:\Users\gr770\DFTF710RC\DFTFringe.exe include/armadillo_bits/Mat_meat.hpp:424
 7# 0x0000000000507AAE in C:\Users\gr770\DFTF710RC\DFTFringe.exe zernikeprocess.cpp:1422
 8# 0x00000000005085A0 in C:\Users\gr770\DFTF710RC\DFTFringe.exe
 9# 0x000000000050C4C2 in C:\Users\gr770\DFTF710RC\DFTFringe.exe
```

gr5, 2023-11-07:
I learned more. I didn't realize it but I had a small "inside outline" in the OLN file.  It was so small I didn't notice it.  This could be a very old bug.

atsju, 2023-11-07:
relevent version : cd52669dd953b2f5b01b3e1cb6947dddbfd0dff1
I just tried with this binary and did not reproduce : https://github.com/githubdoe/DFTFringe/actions/runs/6776965033

Please share the igram and oln you used. It hink you can zip it and put here

gr5, 2023-11-07:
You have to create an inner outline:
after creating outer outline, but before processing the igram, select the "central hole" radio button near the top, then use the mouse to draw a circle inside your igram.  I did about 10% of the total diameter.  I've tried many different sized inner diameters including some only a few pixels across.

Only then will it run the function zapmC().  version 7.0.0 also crashes.

atsju, 2023-11-07:
I understand now. I might try tomorrow. At least you will know if it crashes in debug mode on my computer. It might help a lot if it does.

githubdoe, 2023-11-08:
You are going to find some code that calls zapmC that does not make a lot of sense.  That is the area were I was going to call the anulus zernike generator provided by Mike Peck.  I had that code calling his if the inner diameter  was a greater percentage of the outer diameter by (??  perhaps 30%).  However Mike discovered a bug in his routines that I used.  I killed the if statement switching to the anulus.  But it is still there but will never trigger.  So there is a bunch of unfinished stuff there.   

Try your best to ignore that monster.

githubdoe, 2023-11-08:
Don't get confused like I just did.  There is zpmC and zapmC .   The "zapmC" is the anulus zernike generator.  
I thought I had disabled it but perhaps bad memory.  Instead I see that the if is 

if (obsPercent <= 0.)

I can't figure out why it compares to 0.  At one time I did not want it to kick in except when the ratio was 30% .  Did the 3 get lost?  I don't know.

gr5, 2023-11-08:
So the stack may be getting corrupted because the stack trace doesn't quite agree with logging.  I'll show you the actual code with my temporary added logging I suppose.  Further below.  Here is the logger output:

[2023-11-07 22:28:34.991] [trace] starting zapmC
[2023-11-07 22:28:34.991] [trace] zapmC 1
[2023-11-07 22:28:34.991] [trace] zapmC 2
[2023-11-07 22:28:35.040] [trace] zapmC 3
[2023-11-07 22:28:35.040] [trace] zapmC 4
[2023-11-07 22:28:35.220] [trace] zapmC 5
[2023-11-07 22:28:35.220] [trace] zapmC 6
[2023-11-07 22:28:35.223] [trace] zapmC mmc 0
[2023-11-07 22:28:35.223] [trace] zapmC nrow 281760
[2023-11-07 22:28:35.223] [trace] zapmC nj 8
[2023-11-07 22:28:35.223] [critical] Unexpected issue. Stacktrace:

And the next link shows you the function where it crashes on line 1339. where it calls qr_econ.  First time through the mmc loop.  I don't really understand what qr_econ() does.  I don't understand much of any of this code.  And why does it call different code if there is an inner obstructed region?

https://github.com/githubdoe/DFTFringe/blob/566681e3a15c1b5226a63d47dc61bba51d4ce329/zernikeprocess.cpp#L1293C1-L1343C46

gr5, 2023-11-08:
@githubdoe - I just saw your comments from 3 hours ago.  Just now.  So...  Maybe I can just disable zapmC?  The problem is that it actually works fine when I build with QT Creator (debug or release version).  But if I can just disable it, that would make my life easier, lol.

I suspect there is some kind of memory clobber but when I build it, it doesn't crash because of some kind of "luck".  I partly suspect this because the stack seems like it is damaged when it crashes.  Many of the matrices are on the stack in this zapmC function but I don't understand the standards (which functions expect pass by value for example, when it's okay to delete matrices, and so on).

githubdoe, 2023-11-08:
I don't know if the annulus zernike routines are the one Mike Peck published last as working.  I kind of think not.  In that case we should not provide an annulus option and suffer along with the original which will create wrong zernike values for spherical terms once the obstruction becomes greater than about 20 to 25%.    It would be very nice if that was all fixed and tested correctly however.  That requires a lot of work like getting Mike's latest C routines.  zapmC  stands for the annulus zernike generater in C called by his R programs.  Where ZpmC is the standard zernike generator.

I think that this year we had a user that needed good Annulus zernike terms and there were conversations about it on the group.  Look back and see when you get time.

gr5, 2023-11-08:
Ok then I'm just going to change that "if" statement so the annulus version never runs.  For release 7.1.0 and I'll create an "issue" to fix this with Mike Peck's code at a later date.

Although if you use his code to create the zernike's - and then create the smoothed wavefront based on that - it will work great yet the zernike terms displayed (only in the smoothing dialog) will be different than those created for non-annular zernike's.  And the null won't match properly either.  Which is probably fine.

To do it right all the code related to the null and the zernike's will have to have two different ways of displaying and thinking about the zernike's.  "normal" zernike's and "annular" zernike's.  That's probably a big project.

gr5, 2023-11-08:
@githubdoe - okay I made the change and it's this pull request: https://github.com/githubdoe/DFTFringe/pull/114

**Please review.**  I only changed one line of code plus I edited the release notes in preparation for potential release of 7.1.1.  I still need to test this much more so I won't release it tonight but possibly tomorrow.  Soon.  I just want to spend at least an hour testing it out after this pull request is done.

gr5, 2023-11-10:
fixed in v7.1.1

closed 2023-11-10 by gr5