githubdoe/DFTFringe #118: Building for macOS - QT

Interferometry, 2023-12-19:
This is an issue to track macOS build issues.

The version of the QT framework we use in DFTFringe cannot be built with the latest release of macOS (Sonoma, version 14). I have to use this macOS version so haven't been able to build DFTFringe for a couple of months now.

The good news is that the problem has been fixed in the commercial version of QT. The bad news is that it doesn't look like it will get released into the open source repository anytime soon.

However, the relevant issue(s) in their bug tracker do contain all the patch info required (I think!). Yes, its a bit messy, but I dont think there is a great solution right now.

So, I'll attempt to build a new version in the not too distant future!

Dan

atsju, 2024-08-14:
By this comment I'm just linking issues #120 #24

atsju, 2026-08-21:
closing as done in #354