githubdoe/DFTFringe #120: Need to document how to build

atsju, 2024-01-22:
Currently there are 2 files documenting how to build:
https://github.com/githubdoe/DFTFringe/blob/master/How%20to%20Build%20DFTFringe.pdf
https://github.com/githubdoe/DFTFringe/blob/master/buildingDFTFringe64.txt

They are 2Y old and not accurate. I would like to write a new documentation with current versions of every dependencies in markdown format to be maintainable.

I though one could read the github workflow files to do the same as automated build and not duplicate the "information" but this is clearly not user friendly enough. So a write documentation dedicated to human will be made when I have some time.

It needs to be done for 

- [ ] Windows
- [ ] Linux
- [ ] Mac @Interferometry help will be welcome on this as I don't own a Mac

gr5, 2024-03-06:
I have my own document which I've posted on groups.io before but I never checked it into git.  These are my own notes of how I do it using QT (which is free).  Through my process you get the QT debugger and you can set breakpoints, step through code, examine variables and all that.  And edit the "ui" files with a visual editor.

You can ignore 90% of it and just look at the lines that start with "64bit"
[building_dftfringe.txt](https://github.com/githubdoe/DFTFringe/files/14514923/building_dftfringe.txt)

atsju, 2024-03-07:
Thanks @gr5 
You are right and I will add how to debug. 
My method also works for debugging, it's just I describe the full build with all dependencies build.

Do you think it makes sense to add a "reduced" build process that would rely on DLLs from a previous installation ?

gr5, 2024-03-07:
I don't know.  It's worked really well for me.  I thought we should just check in the dlls in a subdirectory so people don't have to make them to build the project.  github could skip that as well.  And they will probably only need to be updated to a newer version maybe once every decade.

atsju, 2024-03-07:
> I don't know. It's worked really well for me. I thought we should just check in the dlls in a subdirectory so people don't have to make them to build the project. github could skip that as well. And they will probably only need to be updated to a newer version maybe once every decade.
> 
> Dale didn't like that idea.

That would make us lose the knowledge on how to build. It's better to know how to do properly and add a method for "lazy" (or smart) developers than to just add the DLLs to Github. Also there would be licensing issues.  
I will add your lazy/smart method when I write the doc.

Note that GitHub does not really build the DLL each time. I made use of a cache and when nothing changes, the cache from previous builds is reused.

gr5, 2024-03-07:
> That would make us lose the knowledge on how to build.

Good point.

Interferometry, 2024-07-06:
See https://github.com/githubdoe/DFTFringe/pull/135#issuecomment-2211826339 regarding the macOS build.

closed 2025-08-14 by atsju