githubdoe/DFTFringe #131: Unable to save wavefronts

ConradOptics, 2024-02-29:
I am using v7.0.0
I am using the carrier fringe capability and the software is working fine for calculating wavefronts  
When I try to save a wavefront it locks up and does not save the results.  
I have tried saving files with different array sizes thinking that might help but it does not.  
I use the files/save wavefront/s and after a few seconds the software does not respond.  

windows 10Pro
version 22H2
dftfringe v7.0.0
QT version 5.15.2
compiled with MinGW-w64 64bit 6.0 GCC 8.1.0
open cv 4.6.0
QWT 6.1.6, armadilo 12.6.1

atsju, 2024-02-29:
Hi,
Thank you for reporting with every detail.

Just to simplify investigation, please confirm the defect is still present in latest release (v7.1.2 as of today) [available here](https://github.com/githubdoe/DFTFringe/releases)

Also, If you could please share a minimal but accurate process to reproduce I would be grateful as it would save much time to developers.

gr5, 2024-02-29:
There are 2 bugs with saving wavefronts.  One is it won't work if there are any unicode characters in the folder name.  But in that case I think it says something like folder not found or something.  I forget.

The other bug is if you save to a network drive it's slow.  Extremely slow.  It can take 3-10 minutes to save.  Make sure you save wavefronts only to local drives.  This is on the list of bugs for me to fix.  I should probably up the priority as it bit me just this week.

ConradOptics, 2024-02-29:
OK, when I save to the local drive it works.  The issue was saving to a network folder.  
SO perhaps we can close this,

ConradOptics, 2024-02-29:
thanks for all your work!

atsju, 2024-02-29:
thanks all for prompt response and solving :)

closed 2024-02-29 by atsju

ConradOptics, 2024-02-29:
Just so you know, I am a PHD in optics from the U of Arizona and have 20 years of experience on JWST.  I am an interferometry and optical test specialist.  I work for a small non-profit and your software is great.  

Question--- is the wavefront smoothed in the data or is it truly full point by point data?

ConradOptics, 2024-02-29:
My name is Conrad Wells and I would like to be able to interface with you if possible to answer questions like this.  I will also reference your work and software when I publish my results in SPIE.

atsju, 2024-02-29:
Nice.
If you have any question like this, feel free to search and post at https://groups.io/g/Interferometry
The original developer Dale is more active there and there are plenty of other experts there. On GitHub we are mainly 3.
@githubdoe and @gr5 know much more than me on all aspects. I'm only helping a little bit for the software development.

githubdoe, 2024-02-29:
Hello Conrad.  I'm the developer of DFTFringe.  I have a few other software helpers here that have been responding to you.  I'm glad you have found DFTFringe useful for your work and look forward to helping you when I can.  Up until this point I have usually  discuss DFTFringe with folks on the Groups.io group as Jullian has stated.  DFTFringe saves each point without modification or smoothing.  It can be told to smooth and then the displays use that smoothing.  The number of sample points can be specified on the DFT analysis tab usually us ATM's use 640 to 1000 square array of points.

I'm not a trained optical engineer but instead an electrical engineer with and extensive back ground in software.  I got most of my optics knowledge from the group and analysis of David Rowe's FringeXp fringe analysis software.   All done 10's of years ago.  I'm willing to answer any questions you have but prefer to do so on the interferometry group since it is usually of interest to those users as well.