githubdoe/DFTFringe #132: Saving on network drive is incredibly slow

atsju, 2024-02-29:
**Describe the bug**
As mentioned [here](https://github.com/githubdoe/DFTFringe/issues/98#issuecomment-1762958562) and in #131 DFTFringe is incredibly slow when saving on network drive

githubdoe, 2024-02-29:
I find many posts about that problem which is more than likely a Qt problem.  I know there are other apps based on Qt that I can't imagine have that problem.  So there must be away around it.  They say it is caused because Qt breaks the write up into very tiny byte steams.  That all I know at this point.  I realized I do have a way to test it here with a networked drive and it is taking forever on a 1000 x 1000 wave front.  So far more than 15 minutes.

gr5, 2024-02-29:
I think the fix isn't so hard.  We just need to buffer it ourselves and send it out through some other method like fprintf().  It's a bit of a mystery why it is so slow as I think Dale you are already using a buffered stream.  But that is the cost of using other people's libraries - sometimes it "bytes" you.

In windows, the file is created almost immediately and you can right click and look at properties and watch the file size grow slowly over many minutes.

gr5, 2024-08-02:
Fixed.  Will be in next release > 7.2.1

closed 2024-08-02 by gr5