githubdoe/DFTFringe #136: Read wave fronts dialog does not allow multiple file selection

githubdoe, 2024-03-07:
I may have caused this.  I will investigate.

Looks like the enable of multiple files is missing.

I think there should be a line:
dialog.setFileMode(QFileDialog::ExistingFiles);

after the dialog is created in Mainwindow.  When I add that line it works.

githubdoe, 2024-03-07:
Yes I caused it.  in Mainwindow.cpp 
dialog.setFileMode(QFileDialog::ExistingFiles);
should be added back in after line 615.

I was trying to get the last path correct so that dialog would open up more or less in the correct directory when load wave fronts was selected.  That required playing around with what default path got selected.  I think that part is right but I deleted the multiple file selection enable as well by mistake.

I don't have latest code so I'm afraid to make the change myself until I can update my local repo.  Which because my status is on a branch and I'm never sure of the magic words needed to get myself up to date.  Maybe someone else can add that line back in.

githubdoe, 2024-03-07:
I gambled and pushed my  mainwindow mod to annulus.  Don't know if that helps or hurts.

atsju, 2024-03-07:
Hi Dale,
I made a branch, cherry-picked your commit and opened a PR from your fix. See #137

atsju, 2024-03-07:
For now, you should probably do following:

This will get you to clean ans latest commit of master. Works always. Doesn't delete anything you comited on a branch (except if you commited on master locally)
```
git checkout master
git fetch
git reset --hard origin/master
```
Then to work a a new feature
```
git checkout -b DOE/feature
git push -u origin DOE/feature
git commit as usual
git commit as usual
git push
git commit as usual
git push
...
```

PR #137 (2024-03-07): restored ability to select multiple wave fronts
fix #136
closed 2024-03-07 by gr5