githubdoe/DFTFringe #140: Missing DLL for windows 7 from version 7.2.0

atsju, 2024-03-07:
See https://groups.io/g/Interferometry/message/34506
and https://groups.io/g/Interferometry/message/34514

I propose not to fix as Windows 7 is more than end of life now. In the long term it would probably cause more harm than good as we would be tied to old versions of tools and ultimately would not be able to address newer issues on supported OS.

Oleg has no problem with this as he has access to a more recent computer. So nobody is impacted.

gr5, 2024-03-07:
Let's add it to the release notes:

"fist version incompatible with windows 7"

gr5, 2024-03-09:
By the way, that dll has something to do with "touch".  Like if we wanted the installer to run on an ipad, an iphone, or similar.  But anyway we don't plan to support a windows 7 installer so I'm closing this for now.

closed 2024-03-09 by gr5