githubdoe/DFTFringe #142: Crashing

githubdoe, 2024-03-08:
I don't know the exact sequence but it has crashed several times when I try to use the simulation to create a wave front for a normal circular surface  16 inch F3 cc -1.  When I create one wave front then change the amount of correction and try to create another it will crash.   Sometimes when I change the desired cc to 0 it will crash.  Sometime when I change the desired cc back to -1 it will crash.  Sometimes when disabling spherical and recompute it will crash.

While try under debugger to get more info.

I was playing around trying to mimic a ronchi of the nulled wave front where spherical was not present.

githubdoe, 2024-03-08:
Tried my local build version.  It too crashes when I make several changes like I described above.  So probably not related to any github build issue.

githubdoe, 2024-03-08:
Quick update.  It crashes in debug and shows it is in Zernike process make surface from zerns.  So it probably will not be too hard to find.  Probably the work I did modifying the init grid and the like routines to create an array of points only inside the mirror outline.

githubdoe, 2024-03-08:
I found it.  Darn.  It is caused by my creating the rhotheta points but using the size of the matrix number of rows instead of the size of rhotheta.   I fixed it earlier in another location.  But there is at least this point also needing it.  So now I will update my archive and then make the changes as described by you all.

githubdoe, 2024-03-08:
Ok I think I'm done and have pushed all.  Also updated annular attribution in the about info.

atsju, 2024-03-08:
Nice branch and naming convention. I opened the pull request for you.

gr5, 2024-03-08:
Does this also fix the crashing when you check/uncheck the annulus box while a wavefront exists?

closed 2024-03-08 by gr5

githubdoe, 2024-03-08:
Probably.   It does not crash for me at least but don't know about before.  I can see where this might have caused it.  I tried to find other places where it would happen and found none.  I searched for every where m_zerns was used to control a for loop but rhotheta was actually accessed.

gr5, 2024-03-08:
Dale I found another crash.  See my email to you earlier today or just go here: https://github.com/githubdoe/DFTFringe/issues/144

PR #143 (2024-03-08): Fix crash in Zernike process
Fix #142