githubdoe/DFTFringe #144: another out-of-matrix crash

gr5, 2024-03-08:
**To Reproduce**
Steps to reproduce the behavior:
open DFTF, in mirror config click annular box and set to 50%.  Nothing else matters in mirror config.
simulate igram, settings don't matter but I did 600 pixels, click ok
in igram window click "done" (no need to set inner outline), click "compute surface"

now go back to "mirror config".  Uncheck annulus, click ok.  Wait a few seconds for it to process.  now go to mirror dialog, check annulus and click ok. 

You can see the progress bar slide across and it crashes part way through - here's the line of code where it crashes in ZernFitWavefront()

For my 600X600 igram and DFT, it crashes when i reaches 211288.  m_rhoTheta.n_cols is unfortunately larger at 277976.
[image]

gr5, 2024-03-08:
Dale seemed to find the last issue pretty quickly.  I'm hoping this one is easy as well.  It seems some things are cached and not properly recalculated?  Do I have that right?  Is it possible to encapsulate the problem in a class where the class itself keeps track of if things need to be recalculated?  I'm not sure what the issue is.

githubdoe, 2024-03-08:
I tired to make initgrid smart and not always run if it did not need to be.  It is still to stupid.  We keep finding cases where it should run but does not think it does.  I'm inclined to just remove that if need be code.  However then we may discover why I made it in the first place.  

I don't have time at the moment.  But you can play with it if you do.  You will see an if with a lot of conditions.  That is the one.  In zernikeprocess::initgrid().   There is also a variable that others can set called needsinit.  That is how I fixed the last one when the caller wants to force an init.

gr5, 2024-03-08:
Ah.  I'll look at this...

gr5, 2024-03-08:
Okay.  Fixed.  Please review PR #145

PR #145 (2024-03-09): Fixes a crash for when you switch from annulus to regular and back again
Fixes #144 

I almost made it set m_lastusedAnnulus to false but that would also be bed as initgrid still needs to recalculate zerns if called while mirror is non-annulus.

This should fix the problem.  I mean it passes my test.  If approved I will create a release and do more testing of annuluar and normal igrams.
closed 2024-03-09 by gr5