githubdoe/DFTFringe #147: mirror config bugs

gr5, 2024-03-12:
**To Reproduce**
Steps to reproduce the behavior:
In mirror config make sure "units in mm" is checked, then load a json file version of an existing mirror config preferably with annulus enabled in that file.

  note that after loading, values are in mm but the "units in mm" box got unchecked.   It shows mm units but checkbox is cleared.

Now if you check the box things go weird and in particular the annulus diameter gets 7X too large.

**Expected behavior**
units shouldn't change.  Values loaded should match state of "units in mm" box.

**Desktop (please complete the following information):**
dftf 7.2.1
windows

githubdoe, 2024-03-13:
The bug is that .json code is reading the units value from the .json file.  However old config never read units config from the file but instead always expects units to be stored in mm and never in inches.  The .json read code should be altered to just set the units variable and check box to true.  Making sure first that writing .json file also stores  units in mm.

githubdoe, 2024-03-13:
I think I fixed it and did a push.

gr5, 2024-03-13:
1) you didn't have to fix it.  I could have done it.  Just letting you know.
2) I looked at your code.  Looks good.  I'll test it.
3) next time please push your changes in a differently named branch so we can do the whole pull request process.

atsju, 2024-03-13:
Let me know if you want me to cherry pick or rebase Dale commit in a fresh branch.

gr5, 2024-03-13:
Your fix works fine however...

I should have listed the other bugs but I was going to investigate them and fix them at the same time.  I said "bugs" for a reason.  but I should have explained that.

Anyway there are 2 more bugs.  

One is that annulus diameter caps out at 1 meter no mater what.  Someone might want to use this software on a 2 meter f/5 mirror with 51% annulus but dftfringe refuses.

Another is that if you are in inches mode when you read in the file, the annulus diameter appears to be sometimes the inches value and other times 4X too large.  Now I can't repeat quite exactly.  When you switch to inches mode, it actually switches the diameter to mm mode (if it had been inches but not if it had been mm).  Then when you switch to mm again it multiplies by *another* 25.4 and clips at 1000mm.  

I think I got the description close enough to repeat the bug - basically load file while in inches mode and then toggle mm/inches a few times.  This whole thing is weird because the annulus diameter isn't stored in the json file but seems to be calculated on load.  Which is a good thing but should prevent bugs like this.

Anyway I'm not asking anyone to fix this.  I don't think this is an urgent or serious bug.  I plan to fix this some day most likely but not today.  If I do decide to take this simple little bug on I will post a comment here first.

I suspect the bug is that when loading a value into one of the fields, there is a change event which runs before all the data is copied from the json file.

gr5, 2024-03-13:
> Let me know if you want me to cherry pick or rebase Dale commit in a fresh branch.

I think we can just let this one slide as is.  At least it's documented who did it and when.

atsju, 2024-03-13:
> > Let me know if you want me to cherry pick or rebase Dale commit in a fresh branch.
> 
> I think we can just let this one slide as is. At least it's documented who did it and when.

No problem to let it slide as is. 
For information, when I do this, author and date is unchanged. I become co-author I think but main one is original one. Cherry-picking is moving a whole commit not just the code changes. 
See #137 and #125

gr5, 2024-03-13:
I know but it's fine the way it is.

githubdoe, 2024-03-14:
Sorry I had forgotten the exact process and just fixed the master.  I was going to let you do it but then after writing up what I thought I thought I should at least test it.  Then since it worked I decided to update the master.

gr5, 2024-07-04:
It's a miracle: I have some free time! 🙂

I'm going to look at this.

gr5, 2024-07-08:
Dale, what is that line of code for?  I assumed it was for updating progress bars but when I commented it out and tested versus the latest release there was no difference in progress bar activity.

I tried a 1000X1000 DFT (to make it slower) and tried things like hitting "recalculate" button and defocus stuff and just processing igrams.  Also changed to annulus type igram and back (annulus igrams definitely show a progress bar).

There was no progress bar other than for annulus and that line of code doesn't seem to be used for annulus.  so:

**What is that line of code for anyway?  If not for progress bars?**

gr5, 2024-07-08:
oops.  last 2 comments on this issue were for a different issue.  I'll repost there.

closed 2024-07-09 by gr5

gr5, 2024-08-02:
Fix will be in next release > 7.2.1