githubdoe/DFTFringe #150: Will multi-language support be added in the future?

murenshx, 2024-04-03:
I would like to ask if there are plans to add multi-language support in future updates? This feature will greatly improve the user experience, especially for those who want to interact with the software in their native language. 
    A friend of mine recently had some difficulty trying to change the language in DFTfringe. It seemed that the current version of the software did not support this feature.
    Thank you and the team for your efforts on the open source software name, and hope that you will consider adding multi-language support in future versions. 
    Thank you!

githubdoe, 2024-04-03:
I don't plan on doing that but maybe someone will.  There are hundreds of text strings that need to have special codes prefixed to them in order for that to happen.  It is not just adding the translation tables that is necessary.  Then of course there is the amount of space the new text takes up on the screen that can be problematic with column based tables and graphs.  I imagine it to be a huge task.

gr5, 2024-04-03:
Well tool tips are unlimited size so we can say which strings need to have a maximum length and which are associated with a tool tip.  That way the short strings can be abbreviated.  I'm not sure how one does this.  I'm from the old-school windows programming age when we were supposed to use "resource files" to store all the language data.

I don't know if there are any good helper functions to make things well organized and still keep the code readable (hiding the strings makes everything more difficult to understand what the code is doing).

githubdoe, 2024-04-03:
Qt documentation.

[https://doc.qt.io/qt-5/i18n-source-translation.html](url)

I think the original text is used as a key into a translation table.  I only read a tiny portion.

githubdoe, 2024-04-03:
Well that link does not seem to work here.  Don't know why.

githubdoe, 2024-04-03:
I seem to remember that someone did translations for French.  So part of the structure must already exist like the resource.

gr5, 2024-04-04:
Link works fine if I cut and paste it.  I read what was there earlier today.

gr5, 2024-08-02:
I spent quite a few hours looking into this and looking at how other applications deal with this.  We would want to add in "settings" the ability to choose a language.

There is an XML ".tr" file created for each language.  It has the english original language and you can type in the translation.  There is also a nice tool in qt creator called "linguist.exe" which makes it easier to edit the ".tr" file.  Inside the tr file is information about what line of code the untranslated string is located on and there is an update feature to update the tr file to the new line numbers when code is changed.  Another step that we need to do when we make edits to code!  Maybe it can be automated - not sure.

There is an easy function that can be called at runtime to change the language on the fly although you have to reopen dialogs after so it's best if you restart DFTF after selecting a new language.

Dale already created a ".tr" file for french language but nothing is translated yet.  Someone would need to enter the translated text for each chunk of english text.  More code has to be written to select a language and to store the selection and I believe the tr file needs to be moved into a ".qrc" resources file.  The project file needs a few changes.  But overall it's not super hard.

So the big question is:
**Is there any DFTFringe user out there who is wants to translate DFTFringe texts to another language?  Without someone happy to translate I don't see much point and moving forwards.**

atsju, 2024-08-02:
Thank you @gr5 for the work.

I have some followed issues opened here for example https://github.com/jetperch/pyjoulescope_ui and what I see is people complaining about bad German translation for example.   
Personally I always use such tools in English to avoid bad translations.  
Seeing the limited (compared to widely used apps) number of users I would recommend sticking to English and put our development effort on other issues. Translations will always miss a language, be bad quality and take time to maintain. Some technical vocabulary is never translated anyway.

gr5, 2024-08-02:
I suspected as much.  I don't think we have enough users for translation.  If we had 10 foreign language people on the development team then it would make more sense.

But @murenshx - if you have the energy to translate let me know.  There's currently 710 things to translate in the ".ts" file and I suspect there are many more that aren't in the file as I have to add "tr()" function around strings that aren't already in the "ui" files (the "ui" files have things like the text on a button).  So I would guess maybe 3000 things to translate.  And some of them will be difficult as they need to fit in a small space.

githubdoe, 2024-08-02:
That has always been one of the lowest priority tasks I can imagine.  I remember years ago looking into it but the task was too big with little payback.  Yes there are so many things that need the tr() added and as you said the spacing becomes an issue.