githubdoe/DFTFringe #151: Defocus adjustment causes a crash

githubdoe, 2024-05-15:
Create a wave front.  This case for a 16 F3 mirror with 99 percent correction a 5 defocus.
Press the defocus adjustment control on the lower right side.
Try various settings in particular more than 10 wave of defocus.  DFTFringe will crash as you change the settings.
Also looks like the mm display does not update sometimes as well.

gr5, 2024-05-15:
If you work on this yourself, please grab the latest "master" before starting.  Last time the merge was difficult.

gr5, 2024-05-15:
I will try to duplicate in the next few days.  It worked fine for me last time I tried but that was probably before annulus version.

githubdoe, 2024-05-16:
The crash happens as you increase the defocus to tens of waves somewhere past 40 or so I think.  Most will never need such a large defocus in reality.  I was trying to see how many waves of defocus it would take to change the ROC of a 16 F3 from 96 to 95.5 inches.

gr5, 2024-07-09:
fixed in pr #156

gr5, 2024-08-02:
Fix will be in next release > 7.2.1