githubdoe/DFTFringe #152: annulus config is not recomputed

atsju, 2024-05-20:
**To Reproduce**
Steps to reproduce the behavior:
1. configure a mirror with annulus
2. import some pictures, outline and all the usual
3. "oh I forgot to change the config from previous mirror. let's remove annulus"
4. configuration-> mirror-> uncheck annulus. OK
5. Current wavefront is updated. Cool :)
6. look at next wavefront => It still shows annulus hole
7. open and close config window is enough to recompute the wavefront but it shall not be necessary.

**Expected behavior**
I expect that all wavefronts are recomputed when I display them.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: win10
 - DFTFringe version: 6.2.1
 - Build: official GH release

atsju, 2024-05-20:
I see the same for "Gaussian Blur diameter". I see it checked or unchecked but wavefronts are not updated on loading them. 
When any configuration common to all wavefronts is updated they should be all systematically recomputed. Selecting them all fixes the issue but is long to load. Maybe they should all have an associated boolean telling that they need or not recomputation. Once they have been loaded once, no need to recompute unless a config has changed.
OR they get all recomputed each time we display them but this will cause slowness.

gr5, 2024-05-20:
Same thing for checking and unchecking zernike's and then hitting "recalculate".

All these changes only affect the currently selected wavefront.  But when you switch to another wavefront, the checkboxes and values are reflecting the wavefront you were previously on.

To have these changes recalculate all the wavefronts and if there are say 30 wavefronts open - this might be very annoying to the user and take several minutes.

I'm not sure what the best solution is but I feel we need to figure out the best UX (user experience) first.  Probably when you switch wavefronts it sets all the gui elements back to what they are for that particular wavefront (in the mirror config as well though?  when you change the mirror config should there be an "apply to all wavefronts" button as well?)

The way it works now isn't too horrible once you realize it only affects the current wavefront until you hit "recalculate" button.  We could also have a different color for wavefronts that aren't matching the gui and/or a recalculate button next to each wavefront that hasn't been updated.  And a recalculate all wavefronts button in the same region of the GUI.

githubdoe, 2024-05-20:
The current behavior is on purpose.  That is only the selected wave fronts should be affected.  Yes sometimes you want them all so then you have to remember to select them all.  There are many times I don't want them all so this works for that case.

githubdoe, 2024-05-20:
I should add that is the behavior I want because I quite often want to compare wave fronts with different settings.

atsju, 2024-05-20:
Ok I understand your need to have different settings for different wavefronts. This is an important information and we will take that into account. 

However, when you select different wavefronts the UI settings check boxes do not reflect this particular wavefront and this is really bad user experience. 

So for the setting unique to WFT we need to have the UI update it's state when displaying a particular WFT so that state and what is displayed is always coherent. For setting in the "mirror settings" menu, I think this needs to be common to all WFT. Final user (when you are not developing feature or debugging) does not need different mirror settings in same instance of DFTFringe. 

What do you think ?

gr5, 2024-05-20:
That partly works for me:  Save the state of certain things (zernike checkboxes, gaussian amount and checkbox, maybe defocus feature settings) but should we also save mirror config settings and change those as well?  That seems more confusing. The null is indeed displayed on that page.  Do we need annular and null values displayed to the right of the wavefronts so it's obvious which wavefronts have which null?

githubdoe, 2024-05-20:
More than just a debug tool.  I use it all the time when not debugging.  I want to compare different features of the mirror which show better at different settings of things like zernike values (ie. astig) or blurring or.....

In fact I would really love to use different mirror configs on the same group of wave fronts.  However that was too hard to implement.

githubdoe, 2024-05-20:
George asked.  Do we need annular and null values displayed to the right of the wavefronts so it's obvious which wavefronts have which null?

Maybe but what ever is displayed per line becomes the name of the wave front and it's file name.  So be careful.  Also takes up more room.

gr5, 2024-05-20:
No I would put it as a second column to the left or right of where the text is.  Right now we have icons to the left of the wavefront names to indicate which one is selected.  Not sure if you've seen that feature yet Dale.  There should be a way to add 2 more columns hopefully.  one for annulus % and one for null amount.  That should cover all the key settings in the mirror config menu I think.