githubdoe/DFTFringe #155: Add percent correcting graph option to profile plot

githubdoe, 2024-07-07:
A smart user created this percent correction graph feature for DFTFringe.  I'm posting it here to preserve his code.  It is not really usable as is.  It works but I don't like the presentation.  He and I both think it could be made into a special dialog that displays the plot.  Perhaps triggered by right clicking.  There already exists a right click context menu to place it into on the profile plot.

I will add the code here a little later.  I have to go now.

githubdoe, 2024-10-14:
Added code to dalework branch.

githubdoe, 2024-10-15:
Updated and modified this feature to what I think should be the final version.  Ready to try by others if desired.

gr5, 2024-10-15:
I'll play with it tomorrow or the next day.  I should have time tomorrow.

gr5, 2024-10-16:
looks pretty good.

- When "show correction" is unchecked (circled) it should close the dialog.
- I think "always on top" should be checked by default - particularly for new users and people who aren't great at windows and don't know how to alt-tab back to that window if they click on something
- Weren't you going to display an average for the zones?  You created the zone lines which looks nice but shouldn't you say how much each zone is over or under corrected?
- "percen min" should be "percent min"
- There's some gray text that says "help" but doesn't seem to do anything if I do alt-h or click on it

[image]

gr5, 2024-10-16:
Every 50% you have dark blue lines.  Every 10% you have black dotted lines.  I was going to say just the dotted but really all those blue lines and dotted black are visually cluttered.  I think if you went for a lighter blue and grayer black that would still allow you to read off the data visually but not be quite so visually cluttered.  The vertical lines every 10mm also.

The red lines are not too much.  And are hardish to see but will be easier if you lighten the others.  The primary blue line that shows percent correction looks fine to me.

githubdoe, 2024-10-16:
Hover over the help.  It should pop up a tool tip.  
I think adding percentages to each zone is too much clutter since one can see where the zone crosses the line.  
Not intending to add an average of the zone.  The mirror makers that have commented on zones most think that is a waste.  Giving a low resolution view of the data.

Watch  my video I posted on that thread about defocus.  That is what the plot looks like on my 4k screen.  I don't have a low resolution screen to see.  I need you or someone to adjust the values for that and then let me see what it looks like on mine as well.

githubdoe, 2024-10-16:
Here is a pic of the plot on my screen.  Of course my 4K screen is not the same.  Just to let you know why it looks so different on your screen.  I have never figured out a fix for screen resolution in Qt.  I have looked on the net many times.
[image]

githubdoe, 2024-10-16:
It has crashed several times now as I adjust defocus.  Can you look to see if the code I have in dalework has your changes to prevent that?

githubdoe, 2024-10-16:
And of course it does not crash if I have debug turned on.  At least not yet.

githubdoe, 2024-10-16:
I will comment on things I find as well.  Those I intend to fix for sure.
Selecting other profile options like single profile or all wave fronts should clear percentage check box and close it.

gr5, 2024-10-17:
> It has crashed several times now as I adjust defocus. Can you look to see if the code I have in dalework has your changes to prevent that?

Yes, you have it.  The change is here:

-        QApplication::processEvents();
+        if (!m_bDontProcessEvents)
+            QApplication::processEvents();

The basic problem was (is??) that zernikeProcess::rhotheta() can get called a second time from within.  This causes a mess and an immediate crash as the code is not re-entrant.

 Maybe just comment that out - the lowest line of code above and see if that fixes the crashes you saw.

githubdoe, 2024-10-17:
Than you George.  I was able to git it to crash even in debugger.   The last place it seemed to show was an out of memory error when trying to allocate for something in the spin widget processing.  I will work at it.   

I can get it to crash by continually pressing the down arrow of the defocus spin widget until it needs to change the multiply widget.    Maybe I have created a tight loop with one spin widget triggering the other.

githubdoe, 2024-10-17:
There is wrong interaction between the controls on the defocus dialog between the multiplier the value and the slider.  I expect that can generate an infinite loop when one emits a signal causing the other to emit back to it.    I will work on it.

gr5, 2024-10-17:
You asked for what it looks like on a smaller display.  This is 1920 X 1024.

The background scales are just much too contrasty.  Black dots should be gray.  Blue lines (other than the plot itself) should be lighter.  Red zones is fine.

[image]

githubdoe, 2024-10-18:
Yes that looks ugly.  I will work on it.  The defocus dialog field interactions were a complete mess and indeed created an infinite loop.  I have fixed it.  Now the multiplier does it job whereas before it was intermittently  applied and wacked by the other controls.

gr5, 2024-10-18:
You may already know this but each gui element has 2 events in QT.  Something like on_text_edited() and on_text_changed().  Well a lot more than 2 but the thing is: one of them is called only when the user edits the data and the other is called additionally when you change the value through code e.g. set_text().  If you use the wrong event then you can get unexpected looping as you say.

This is a QT only feature I believe (not part of windows) because I don't think I've ever seen it before.  This seems to be true of every gui element that can change such as buttons, text boxes, scrolls, drop downs, check boxes and so on.

githubdoe, 2024-10-18:
I used on value changed (value)  to get changes when a user changes something.  For the Defocus fields that means some need to also call that function to make related fields change.  However to keep those changed controls from sending out they changed creating a loop you use the blocksignal(true) command.  I forgot to put the blocksignal command in the code.  In addition there was some very wrong code to update related controls.    

That dialog has several ways to change the defocus value.  One slider for fine adjustment and spinners for larger adjustments.  Then the multiply control to adjust the size of value that the slider affects.    It all works now as I intended.

gr5, 2024-10-21:
I mentioned about the gray "help" and I couldn't figure out how to use that until you told me to hover over it.  But if I couldn't figure it out then other's will have trouble.  So please either:
1) change the text to "help (hover mouse)" or
2) Make it also clickable (I assume this is much more difficult) to see the help text.

githubdoe, 2024-10-21:
On my computer just moving the cursor over the help like you where about to press it causes the tool tip to display.  I guess because it is so fast. 
 Yes making it a button then causes one to have to create a dialog with the text and then the user must press buttons to close it.  Or create a window like a tooltip.  So I just used the tool tip itself.  Many things in DFTFringe already have a tool tip.

gr5, 2024-10-22:
Hover help just doesn't work well for me.  This video explains the problem.  I tried clicking, double clicking, right clicking.  Nothing worked because... well watch the video:

https://www.dropbox.com/scl/fi/swslrh2cwdktsl846ofpw/show_correction.mp4?rlkey=xu7ybd9jqe24zb8k738ha5xpx&dl=0

I know all the other controls have hover help for example in mirror config.  And that *does* work for me.  I expect hover help there.  It's consistent.  It's a windows standard (hover help).  But it doesn't work for that gray text.

gr5, 2025-04-06:
I see you pushed another version to github in the dalework branch.  I should be able to take a look today.  I'll see if I can recreate your crash in ronchi/foucault as well.

gr5, 2025-04-25:
If I click the "show correction" button when there is no wavefront created yet, DFTF crashes.  Could you disable that button when there are no wavefronts?

[image]

gr5, 2025-04-25:
By the way, all of the other checkboxes do not crash DFTF if there is no wavefront

gr5, 2025-04-25:
Okay - another bug that should be easy to fix.  If I use the "save zones" buttonto save a zone file, then change the zones, then load the zone file back in, it only updates the number of zones but it still shows the old zones, not the ones from the file.  If I change the percent min or max (y axis range controls) then suddenly I see the old zones back from the zone file that was just loaded.

So somehow the display isn't being updated after loading the zone file (only the qty of zones).  Also the table at the bottom gets cleared out (all cells empty of numbers) until I adjust the percentage a tiny bit and then it all works.

gr5, 2025-05-17:
done.  Implemented in version 7.3.1

closed 2025-05-17 by gr5