githubdoe/DFTFringe #158: DFTFringe oddness under Linux running over ssh vs local

crw4096, 2024-08-08:
Hello,
I posted this in the [Interferometry group](https://groups.io/g/Interferometry/topics?index=107721401) and they suggested here would be better.
Since that posting, I can add that I deleted and recloned the repo and did the compile locally instead of over ssh, but the issue persists. Any ideas?

Thanks,

Charles Wright

-------------------------------------------------------------------------------------------
Hi all,
Been using DFTFringe for a while with Windows in a VM, but when I saw the README explain how to build under Linux, well, that is certainly preferred by me.
At any rate, I run Linux Mint, and could only install the correct dependencies under Mint 21.3. This built "just fine", as one would hope. But now the weirdness.

 * My *desktop* machine has Mint 20.3 and could not easily install the dependencies
 * My *laptop* machine has Mint 21.3 and everything installed and built fine (one compile warning appeared twice for `/usr/include/qwt/qwt_plot_layout.h:84:51`)
 * I ssh to the laptop from the desktop to run DFTF; this works *just perfect* and I never expected otherwise
 * Yesterday, I only had the laptop and wanted to run DFTF. This failed! No window appears and the program hangs until I control-C. See output from DFTF below.

This is just so strange. I know that LM 21.3 does not use Wayland by default (phew) but why would it work over ssh yet not work locally? BTW, I'm working with the latest pull on the master branch.
 
At any rate, really love using DFTFringe to test my mirror as I continue to work it. Nice to be able to compare what I see with a knife edge tester with the measured wavefront error.
 
Best,
 
Charles Wright

-------------------------------------------------------------------------------------------
Output in terminal window using just laptop (no DFTF window):
```
$ DFTFringe 
DFTFringe:180562): GLib-GObject-WARNING **: 09:09:54.218: cannot register existing type 'GtkWidget'
(DFTFringe:180562): GLib-GObject-CRITICAL **: 09:09:54.218: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed
(DFTFringe:180562): GLib-GObject-WARNING **: 09:09:54.218: cannot register existing type 'GtkBuildable'
(DFTFringe:180562): GLib-GObject-CRITICAL **: 09:09:54.218: g_type_interface_add_prerequisite: assertion 'G_TYPE_IS_INTERFACE (interface_type)' failed
(DFTFringe:180562): GLib-CRITICAL **: 09:09:54.218: g_once_init_leave: assertion 'result != 0' failed
(DFTFringe:180562): GLib-GObject-CRITICAL **: 09:09:54.218: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed
(DFTFringe:180562): GLib-GObject-CRITICAL **: 09:09:54.218: g_type_register_static: assertion 'parent_type > 0' failed
 ```
Output in terminal window using ssh to connect to laptop (works just fine - window appears and is responsive...):
```
$ DFTFringe
[2024-08-04 13:47:41.335] [info] 

-------------
[2024-08-04 13:47:41.335] [info] spdlog info
[2024-08-04 13:47:41.335] [warning] spdlog warn
[2024-08-04 13:47:41.335] [error] spdlog error
[2024-08-04 13:47:41.335] [critical] spdlog critical
[2024-08-04 13:47:41.336] [critical] This is a demo stacktrace:
 0# 0x0000559226F97891 in DFTFringe
 1# 0x00007FA123A43D90 in /lib/x86_64-linux-gnu/libc.so.6
 2# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
 3# 0x0000559226F9BF65 in DFTFringe
[2024-08-04 13:47:41.393] [info] DFTFringe MY_AUTOMATED_VERSION_STRING started
[2024-08-04 13:47:41.712] [trace] Log level changed to: Info
```

atsju, 2024-08-08:
Hi, 
Sorry, this will probably be difficult to debug remotely as we cannot reproduce. 
Have a look at these answers dealing with gtk version: https://forum.qt.io/topic/111570/pi-4-qt-5-with-open-cv-4-1-app-no-gui-display-problem-with-gtk/3

crw4096, 2024-08-08:
Thanks - that link was enough to get me pointed in the right direction. The file `$HOME/.config/qt5ct/qt5ct.conf` on the laptop turned out to be the place where setting `style=gtk3` did the trick. Oddly enough, the same file on the desktop machine is set to `gtk2` :man_shrugging:. Anyway, it's good now. 

Charles

atsju, 2024-08-08:
Wonderful 👍 
I will close the issue as it's solved.

closed 2024-08-08 by atsju