githubdoe/DFTFringe #161: Unnecessary 1 second delays in Batch process slows down Bath process and can probably be removed.

githubdoe, 2024-09-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows, Linux, Mac + version informations]
 - DFTFringe version: [e.g. 6.2, sha76d1f40, ...]
 - Build: [e.g. from GitHub release, from GitHub Actions artifacts, your own build, ...]
     - Dependencies versions: [for your own builds only, add version for QT, OpenCV, qwtPlot, ...]

**Additional context**
Add any other context about the problem here.

githubdoe, 2024-09-19:
In mainwindow.cpp  function batchProcess there are 1 second delays I may have put in for debug so I could make various screens display longer.  Those delays are not necessary and slow down batch process.    In a way I don't understand if one on windows 10 just leaves the batch running it will start to delay 10 seconds after displaying the cropped interferogram.  If you move the mouse over the app that delay is interrupted and process continues.  So just moving the mouse randomly over DFTFinge will make the batch work much faster.  After removing those 1 second delays that behavior does not happen and the batch works quickly as expected.

I have the fix and a fix for creating movies from wave fonts that is also disabled currently.  (AT least in my version of the source.)

I would like to update all of this but once again I'm sorry I don't remember thee github commands to update my source with the latest code while not removing my changes.  I know it involves creating a branch but.....

atsju, 2024-09-19:
`git checkout -b yourBranchName `
`git commit -am "my commit message"`
`git push -u origin youBranchName`

@gr5 if the branch is not up to date, you can eventually rebase the commits. I'm in vacation with no computer access

gr5, 2024-09-20:
Dale has been using a really old version of the source code and I think it keeps getting older.

Dale if you could do a pull on master before you make any edits that would be so very helpful.

githubdoe, 2024-09-20:
"before I do any edits"  Edits have been done for sometime so way to late for that.  I imagine a merge will go well.

githubdoe, 2024-10-12:
I hate git.  I have tried but failed to update my local branch with the remote git master.  I have done what the net says to do.
Which is while on my local branch dalework git merge remote master.  But it responds that it would overwrite some new files that I have (those that I have been adding for the new feature)  and will not merge.

atsju, 2024-10-12:
Hi @githubdoe I have tester your dalework branch and I'm able to update it without errors. This means you have uncommitted **local** changes in you branch. Or that the origin/dalework is not equal to dalework meaning you did not push your local branch.

Anyway what I would recommend is commiting and pushing everything from dalework. After that no more risk of loosing anything. Then You or I or We try to update things correctly.  
I also reiterate my offer to have a teams call with control over your computer to help you live.  
We could also make you a cheat sheet to help with git. We are rarely developping in parallel so keeping things up to date ensures there are never complex things to do.

gr5, 2024-10-12:
>I also reiterate my offer to have a teams call with control over your computer to help you live.
That seems like a very good idea to me.

You can also just use the old fashioned copy paste method to move your files somewhere safe on your hard drive.
There's also a feature I love called "git stash" which puts modified files in a safe spot so you can deal with pushing the current commit and then you can do "git pop" to restore just those files that you were working on.

githubdoe, 2024-10-12:
Ah,  Yes I have not committed nor pushed those new files.  
Yes George I did try copying everything to a backup dir and try to update and then move the new files over.  But I messed up and did not copy all the new files over.   I got confused with a couple of obsolete file names.  I eventually got into a state where it would not compile and I got lost.  So I then just copied the backup and I was back to square one.  I have continued making my changes to my old dalework branch.  Will try the commit and push and then merge soon.

I'm working on the percent completion profile graph.  I like what I have so far.

gr5, 2024-10-12:
Okay, well after this next merge, please start fresh with a fresh copy of the latest file now so that in a few months when you next do a bug or feature you won't be working with code that is over a year old and doesn't have all the latest fixes.

githubdoe, 2024-10-14:
I have completed work on the delay, modifying the surface movie maker, and the percent correction feature.  Perhaps some of these had their own issue topics.  I have merged the older master into my code and pushed dalework up to the remote.  (I hope)

gr5, 2025-05-17:
Released in version 7.3.1

closed 2025-05-17 by gr5