githubdoe/DFTFringe #162: gaussian blur causes weird pixelated edges in simulated ronchi

gr5, 2024-10-09:
Version 7.2.3 but I think it does it in many older versions as I've seen this before

**To Reproduce**
I processed igrams from a real mirror.  I'll include pictures of the issue and a picture of a ronchigram of the real mirror.

I smoothed the wavefront with "tools" "smooth current wavefront" although that's not required to see the problem.

I created a perfect simulated wavefront and the same issue happens: looks bad if gaussian smoothing is on.

GOOD:

[image]

BAD:

[image]

ACTUAL:

[image]

atsju, 2025-11-13:
Great work !

I haven't seen the code but now that you know it best: does it make sense to have a GaussianBlur class that would be called from the 3 locations you identified ? (If there are different blur algorithms then a blur class and specific inheriting classes.)

Just thinking maintainability/abstraction and futur fixes/evolutions.

atsju, 2025-11-13:
> I don't see how it's a class

I'm not OOP expert. Discussion with IA, what I though is named a utility class but it seem best approach is a namespace in our case.

What you described is perfectly fine. I'm fine with utils.h.

I will do the review tomorrow or this weekend

githubdoe, 2025-11-13:
I class usually is used for things that have state it wants to save and thus member variables.  This is just a function and is fine as is.  Some would argue it should have it's own class.  I don't care if it is until or it's own.