githubdoe/DFTFringe #163: Crashes when using the defocus value.

githubdoe, 2024-10-11:
Set mirror size to 20 inches F number to 3.  Created simulated wave front with 0 SA value.  Set value to .5 and the multiplier to 20.  Program crashes.  Darn.

githubdoe, 2024-10-11:
Looks like the problem is in the Zernike process and for some reason the index to rho is out of bounds.  I think this might be similar to one George found and fixed sometime back.d

gr5, 2024-10-11:
Try latest DFTFringe please. :smile:

githubdoe, 2024-10-11:
I was using 7.2.3

githubdoe, 2024-10-11:
But what I was debugging was older.  So the crash is not caused by what I saw.

githubdoe, 2024-10-11:
Since I have a couple of versions of DFTFringe with short cuts on my desktop and the one I thought I used said 7.3.2.  But now when I use that I can't get it to crash.  Soooo.  maybe I did not crash 7.3.2.  I will give it a couple of days and then work at closing this.

gr5, 2024-10-11:
If you do decide to write any more code, please checkout the master branch and do a pull first.  The 2 most likely mistakes one would make in GIT (when about to make edits) would be (1) to create your own branch starting with a different branch from master.  Or (2) to forget to do a pull from master before getting started (making a new branch or making any edits).

githubdoe, 2024-10-11:
Yes but like I said awhile back that road of do this before making changes has long passed that point.  So I have code updated to old code.  But I think the merge will be easy.  I just never remember which GIT magic words will update my branch with changes others have made to the master.  I'm always scared to try any for fear it will remove the changes I have made to my local branch.
I think a pull will do what I want and will work on it soon.

gr5, 2024-10-13:
Or he can just check in and commit his changes and ask someone else to do the merge.

githubdoe, 2024-10-14:
I think I'm done.  I have pushed my dalework branch to the remote.  Now I'm ready to update my files to the current master changes.  You talked about rebase and merge and then commented about rebasing will move....   That whole sentence is mostly a mystery to me.  Don't know what "move changes to the head of the branch means" .   Sorry.

    I have files that I changed, new files added, and there are files others have changed that I don't have.      I just want to update and merge those changes as necessary into my local branch.  Then I can push that back up to the remove branch.

Tonight I tried "git merge master"  boy did that make a mess.  It added text conflicts to many files and some of those where in my new files.  How in the world did it not just leave those alone since I thought master would not have those files.  But somehow it did I guess and not only that they were older versions of my files.  Ok I could correct that but why did I have to.  The worst part is it adds conflict resolution headers into files that can not deal with them like the *.ui files.  That took a long time for me to realize that happened.  Qt creator will not show those but instead try to show the GUI those files create.  But compiles will fail.  To edit them out you have to open them in a text editor once you discover they are corrupted.  

Well now I gun shy once more.  I'm afraid to try any merge command.  Can any of you figure out why that did not work? I give me the solution to accomplish my task.  Sorry if you had in the past but I have lost track of any help like that.

githubdoe, 2024-10-14:
I have discovered that I have a local branch called origin/master.  I think that is the problem.  Since I tried my merge as
merge origin/master  thinking that origin was the remote.

githubdoe, 2024-10-14:
So once I deleted extra branches like origin/master the merge worked I think.  Next will see if it will compile.

closed 2024-10-14 by githubdoe

gr5, 2024-10-14:
You did it!  Good job.  I was very nervous that you were working with code from over a year ago but it wasn't *too* bad.  You were working with code from branch "JST/fix120" which was version 7.2.1 plus a few more changes from March of this year.  Not too bad.  I didn't know what code base you were working from until this morning.

But now your dalework branch has all the latest code changes in it including the fix for the crashes in "defocus adjustment".  Also github has the dalework branch so it's safely backed up.

Phew! :sweat_smile:

gr5, 2024-10-14:
So next question:  Is it time to do a pull request and get all your changes into master or are you still working on the "percent correction" feature?

githubdoe, 2024-10-14:
Lets let it simmer for awhile to see if it has settled.