githubdoe/DFTFringe #164: Deal better with non-round mirrors

gr5, 2024-10-11:
Someone contacted me with a strangely shaped mirror and I wondered if he can process it with DFTFringe.  It's a spherical mirror, not parabolic.  Here's an image below - it's a round mirror cropped on 2 sides.  This is a ronchigram.  I'm not sure if this is his 12 inch or his 16 inch (only one of them is cropped).

Anyway, I was able to process a simulated igram with cropped sides using the region feature but DFTFringe uses the regions in it's RMS strehl calculation.  I want an option (rarely used) that excludes those regions in the final RMS and Strehl calculations.  I'm planning to write this feature but my biggest worry is that someone will turn the feature on by accident so maybe this would be something that is *not* stored in the config file.  And maybe this is hidden in the preferences area?  With a warning to use it only for weirdly shaped mirrors e.g. square mirrors?  

I'm not sure if I can even easily do this feature if they are averaging together multiple igrams - how the hell are they going to rotate 90 degrees and remove astig??  Maybe I haven't thought this completely through?  Maybe removing astig will "just work" even though some of it is simulated outside the cropped region.  But if that's true then they shouldn't need to remove the region for RMS calculations as well.

I don't know - this is a crazy feature maybe, lol.

[image]

githubdoe, 2024-10-11:
I think the problem is that the zernike fit must be over a circular disk.  The disk created by extending the outline past that igram will have no data and the zernike fit will not be any good and will not match the data where the lines really are.

gr5, 2024-10-12:
I wouldn't say the zernike fit is garbage but yes, it's not as good as one would hope.  More importantly, the error is mostly in the region beyond that edge and yet that area is still included in the strehl and rms values.

gr5, 2024-10-12:
Here is an example of a simulated igram (actually a real igram of some mirror - I think maybe the first round robin mirror) but I cropped off the edges using the region tool.  I just want to remove those "smoothed" areas that are masked out from the calculation of the strehl and rms error.

[image]

However I forsee more issues if there is astigmatism.  I don't think you can just rotate the mirror 90 degrees and remove astig by averaging.  So unless you have a fantastic test tunnel it will be near impossible to measure astigmatism.