githubdoe/DFTFringe #167: Improve existing auto invert feature that uses conic constant to decide to invert wavefront

gr5, 2024-10-21:
I have never been super happy about the way DFTF asks if you want to invert the wavefront. Also Yordan asked for the ability to automatically invert related to defocus term as an option instead of the SA term.  For spherical mirrors.

I have an idea how to make this easier for users and also add the defocus method. 

The popup would give you more options: 4 radio buttons with options to invert manually, invert assuming mirror is partly parabolized, assuming inside focus, or assuming outside focus (4 options).  Then it won't ask again until you restart the program or change mirror config settings (if null value changes).  Program will have a 5th mode internally which is to ask the user to choose one of 4 options again but if you choose manual it will stop bugging you.

In addition this setting will be visible in the mirror config if you need to change it and it might have the 5th option to "ask me later".

Also save in .OLN files and when loading OLN files if it doesn't match the mirror configuration then either prompt user or (better) just reset things to "ask user" mode again.  If config was in "ask user" mode and OLN file has a mode then use the OLN mode for this one igram only.

If I don't get any comments or if only positive comments I'll draw this up visually to make it easier to see, understand, and talk about.

gr5, 2025-06-18:
Here is a mockup of my proposed mirror-inverted-dialog. There are many special cases so please read above.
choosing any option makes the dialog go away
dialog comes back if you restart the program or chose to open the dialog manually
dialog comes back if you change mirror config
choice is saved in OLN files but only applied to that wavefront
choice is displayed in mirror config dialog and dialog can be relaunched from there

[image]

gr5, 2025-06-18:
That's what the above "manual" button does.  It disables the feature.

githubdoe, 2025-06-19:
I think that manual button will not show up unless you are exercising a make wave front operation.  So it is not universally available.  It also does not show up if you have already said use auto invert.

gr5, 2025-06-19:
> exercising a make wave front operation
Not sure what you mean.

The above dialog would appear the same times your current dialog appears.  I propose replacing your current dialog with this newer one.  Also this newer one would appear when you process the first dialog whether the wavefront appears to need the inversion or not.

You could launch the above dialog manually - I propose it's launched from the mirror config dialog.  It would show the current state with a single word ("manual", "parabolic", "inside focus", "outside focus" (okay that's 2 words)) and next to that a button to change it.  Potentially you could launch it some other way.

ALSO, if you click the invert button for a wavefront I think it should probably also reset the mode/state to manual.

There's a 5th state of course that is "null" or something.  That 5th state indicates to ask the user the next time they process a wavefront.  This is the start up state.

gr5, 2025-06-19:
I'm also worried about the "parabolized" state/mode.  Your current code looks at the null sign and want's S.A. to match.  Is the null ever positive?  Should I worry about that?  Should the "parabolized" mode have a stronger explanation regarding if the null is positive?  What if people are doing elliptical mirrors - should I use a different word than "parabolized"?

githubdoe, 2025-06-19:
I'm not apposed to it.  I think it could be more complex than you realize.  There are a lot of places where it gets used but maybe they all just call a routine.

atsju, 2025-09-27:
You mean you need to merge master into feature branch. Yes merging 7.4 and then 8.0 into feature is a good idea. Then open PR to merge feature into master.

For the PR merge, can it wait my review October 10th or so ? No joke I'm currently at grand canyon.

closed 2025-11-10 by atsju