githubdoe/DFTFringe #169: How to create an alpha version for people to try out.

githubdoe, 2024-11-11:
I would like to publish my .exe of DFTFringe.  To let people try out if they want.  I don't want it to be an install but instead just a down load of the exe after renaming the old version.  I don't know if it could be stored on github some way or in the groups.io file section.  
Any ideas?

atsju, 2024-11-11:
Hi Dale.

I would recommend doing a release identified as v7.3.0-alpha1 by following the usual process. Letting GitHub generating the installer. 

The installer today permits to install several versions side by side so this is not an issue at all.

gr5, 2024-11-11:
I think the release page can default you to any release we choose in github so that it will default you to the newest non-alpha.

To create a release you have to get your version into master and then tag it.  The tag step initiates the automatic creation of the release.  The tag step is probably the easiest step: two commands - one to tag it and one to push the tag.  The normal command line "push" command won't work - you have to specifiy the tag but you can google how to do that.  To start the release process the tag must start with a "v" and I think it needs to be followed by number.number.number and then evidently it can have anything.

Before any of that you have to get your version into master.  Go to the pull requests page and request a pull request and I'll approve it.  So if you go to pull requests, click the green button, it will show your "dalework" as a suggerstion at the top of the list.  Click on that. 

Before you do the pull request it would be nice if you edited RevisionHistory.html

After the release process is done, it needs to be approved in github - edit the text related to the release and save it.

At some later date we can delete the alpha version out of the releases part of github.

I can do any or all of these steps for you if you want.

If you want to revert any changes in the future, git remembers every edit so that can be done as well.

atsju, 2025-06-15:
I will twerk the regex when back to a computer.

It's better to have the correct regex so we always respect server. To avoid mistakes.

atsju, 2025-06-15:
I just had a look and it should already work. v7.3.2rc doesn't server.

You need v7.3.2-anything like v7.3.2-rc1 for example. Let me know if there is an issue

gr5, 2025-06-15:
oh.  So I need the hyphen.  I just reread that semver specification as well.  Thanks.